ZoneMinder/zoneminder #767: Please remove login:password from camera URL path

arjunroychowdhury, 2015-03-30:
This causes various security issues:
a) The username and password are always exposed in logs for anyone to see (not necessary)
b) If someone is using your current APIs, it does not tie into OPT_AUTH, so anyone who has exposed their zone minder via a public IP and has password protected the login and has enabled APIs will expose anyone to gain access to their cameras without need login password because monitors.json returns the full path+password for all camers

connortechnology, 2015-03-30:
Also, see #695

kylejohnson, 2015-04-11:
I have started work on moving the user and pass fields out of the camera path, but I don't have an ETA for this right now.  We all agree that it is sorely needed.

closed 2018-06-22 by stale[bot]