ZoneMinder/zoneminder #769: CMake: insorrect installation of Perl modules [patch]

onlyjob, 2015-03-31:
When I build Debian package using CMake (_autoconf_ is OK because it uses `INSTALLDIRS=vendor`) Perl modules install into absolute location:

~~~~
Installing /mnt/tmpssd/src/zoneminder/tmp/ZoneMinder-1.28.1/debian/zoneminder/mnt/tmpssd/src/zoneminder/tmp/ZoneMinder-1.28.1/dbuild/scripts/ZoneMinder/output/lib/x86_64-linux-gnu/perl5/ZoneMinder.pm
~~~~

I believe this is happening due to gross disregard to `DESTDIR`. Currently `CMakeLists.txt` is not suitable for build environment because `make install` can not put files into package staging directory. In another words the problem is that `make install` do not put files into directory relative to `DESTDIR` because absolute paths are used when modules are pre-installed into temporary build directory.

To fix the problem I've managed to produce the following patch:

~~~~
--- a/scripts/ZoneMinder/CMakeLists.txt
+++ b/scripts/ZoneMinder/CMakeLists.txt
@@ -23,11 +23,20 @@
 	set(MAKEMAKER_NOECHO_COMMAND "NOECHO=\"1>/dev/null\"")
 endif(CMAKE_VERBOSE_MAKEFILE)
 
 # Add build target for the perl modules
-add_custom_target(zmperlmodules ALL perl Makefile.PL FIRST_MAKEFILE=MakefilePerl PREFIX="${CMAKE_CURRENT_BINARY_DIR}/output" LIB="${CMAKE_CURRENT_BINARY_DIR}/output/${ZM_PERL_SUBPREFIX}" INSTALLSITEMAN3DIR="${CMAKE_CURRENT_BINARY_DIR}/output/${CMAKE_INSTALL_MANDIR}/man3" ${MAKEMAKER_NOECHO_COMMAND} COMMAND make --makefile=MakefilePerl COMMAND make --makefile=MakefilePerl pure_install COMMENT "Building ZoneMinder perl modules")
+# PREFIX="${CMAKE_CURRENT_BINARY_DIR}/output" LIB="${CMAKE_CURRENT_BINARY_DIR}/output/${ZM_PERL_SUBPREFIX}" INSTALLSITEMAN3DIR="${CMAKE_CURRENT_BINARY_DIR}/output/${CMAKE_INSTALL_MANDIR}/man3"
+add_custom_target(zmperlmodules ALL perl Makefile.PL FIRST_MAKEFILE=MakefilePerl
+ PREFIX="/usr" LIB="/usr/share/perl5" INSTALLSITEMAN3DIR="/usr/${CMAKE_INSTALL_MANDIR}/man3"
+ ${MAKEMAKER_NOECHO_COMMAND} COMMAND make --makefile=MakefilePerl
+ COMMAND make --makefile=MakefilePerl pure_install
+    DESTDIR="${CMAKE_CURRENT_BINARY_DIR}/output"
+ COMMENT "Building ZoneMinder perl modules")
+
+# ${MAKEMAKER_NOECHO_COMMAND} COMMAND make --makefile=MakefilePerl
+# COMMAND make --makefile=MakefilePerl pure_install
 
 # Add install target for the perl modules
-install(DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/output/" DESTINATION "${CMAKE_INSTALL_PREFIX}")
+install(DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/output/usr/" DESTINATION "${CMAKE_INSTALL_PREFIX}")
 
 # Add additional files and directories to make clean
 set_directory_properties(PROPERTIES ADDITIONAL_MAKE_CLEAN_FILES "output;blib;pm_to_blib;MakefilePerl")
~~~~

But I hope that with CMake there should be more elegant way to pre-install Perl modules into temporary build location. Please review.

onlyjob, 2015-03-31:
Cleaner patch without comments:

~~~~
--- a/scripts/ZoneMinder/CMakeLists.txt
+++ b/scripts/ZoneMinder/CMakeLists.txt
@@ -23,11 +23,16 @@
 	set(MAKEMAKER_NOECHO_COMMAND "NOECHO=\"1>/dev/null\"")
 endif(CMAKE_VERBOSE_MAKEFILE)
 
 # Add build target for the perl modules
-add_custom_target(zmperlmodules ALL perl Makefile.PL FIRST_MAKEFILE=MakefilePerl PREFIX="${CMAKE_CURRENT_BINARY_DIR}/output" LIB="${CMAKE_CURRENT_BINARY_DIR}/output/${ZM_PERL_SUBPREFIX}" INSTALLSITEMAN3DIR="${CMAKE_CURRENT_BINARY_DIR}/output/${CMAKE_INSTALL_MANDIR}/man3" ${MAKEMAKER_NOECHO_COMMAND} COMMAND make --makefile=MakefilePerl COMMAND make --makefile=MakefilePerl pure_install COMMENT "Building ZoneMinder perl modules")
+add_custom_target(zmperlmodules ALL perl Makefile.PL FIRST_MAKEFILE=MakefilePerl
+ PREFIX="/usr" LIB="/usr/share/perl5" INSTALLSITEMAN3DIR="/usr/${CMAKE_INSTALL_MANDIR}/man3"
+ ${MAKEMAKER_NOECHO_COMMAND} COMMAND make --makefile=MakefilePerl
+ COMMAND make --makefile=MakefilePerl pure_install
+    DESTDIR="${CMAKE_CURRENT_BINARY_DIR}/output"
+ COMMENT "Building ZoneMinder perl modules")
 
 # Add install target for the perl modules
-install(DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/output/" DESTINATION "${CMAKE_INSTALL_PREFIX}")
+install(DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/output/usr/" DESTINATION "${CMAKE_INSTALL_PREFIX}")
 
 # Add additional files and directories to make clean
 set_directory_properties(PROPERTIES ADDITIONAL_MAKE_CLEAN_FILES "output;blib;pm_to_blib;MakefilePerl")
~~~~

Besides this is an example how `CMakeLists.txt` could be formatted in a multi-line style to avoid long lines for easier read, see  #760.

knnniggett, 2015-03-31:
Cmake works fine in a build environment. See here:
https://github.com/ZoneMinder/ZoneMinder/tree/master/distros

Each of these folders are for building packages, and they work fine.
You have to export DESTDIR before calling make like so:
https://github.com/ZoneMinder/ZoneMinder/blob/master/distros/fedora/zoneminder.f21.spec#L87

```
export DESTDIR=/path/to/destdir
make install
```

The following does not work:
```
make install DESTDIR=/path/to/destdir
```

onlyjob, 2015-03-31:
I may not be able to pinpoint the problem exactly but I've built enough packages with CMake build system on Debian to be confident that Zoneminder CMake build system is broken. Debian build system sets build paths automatically by `dh_auto_configure`/`dh_auto_build`.

 Once again, this problem is because under CMake invoke Perl build system to install into absolute location prepended with `${CMAKE_CURRENT_BINARY_DIR}` (which contains absolute path as well) into CMake build directory... As result CMake incorrectly installs Perl modules from build directory. As you can see the fix is to give `DESTDIR` with proper location of build directory to Perl `pure_install` and to drop absolute paths to build directory (`${CMAKE_CURRENT_BINARY_DIR}`) from library installation paths.

knnniggett, 2015-03-31:
We have Debian cmake packaging scripts here:
https://github.com/ZoneMinder/ZoneMinder/tree/master/distros/debian_cmake

Compare what you are doing to that.

onlyjob, 2015-03-31:
I smell denial. ;)

What am I doing is building properly packaged official Debian package.
I tried that sloppy packaging from `distros/debian_cmake` and after necessary corrections to make it work I found that it does _not_ install Perl modules properly.
Perhaps I'm building with newer build system/CMake on Jessie but this problem is real and it screams to be fixed.

knnniggett, 2015-04-01:
Well, what do you expect?  You show up out of nowhere a total stranger and state the sky is falling.  

You've done a lot of making claims of broken-ness but you've provided very little factual information about how exactly you set up your environment, and only now have you even stated what you are trying to accomplish.  You haven't really given me much to go on that says anything other than you tried to build zoneminder and it blew up ....as if I haven't heard that one before. I'm sure you can relate to this.

So for the moment, yeah, I am sceptical, because the facts don't indicate what you describe. We've been building packages based on the current method for about year now without experiencing the problem you've described.

I'm not saying I am unwilling to hear your case, nor am I saying I won't help you.  What I am saying is that I need you to show what is different about the way you are trying to build zoneminder.  

So lets start with the scripts `distros/debian_cmake`, and find out if it works or not.  I'm not asking if you like it or not.  That folder is there to make the Debian package maintainer's life easier so ultimately you will likely get the chance to make it look however you want if you stick around.   If you build a package with those scripts, what is wrong with resulting package?  What files (Perl modules) are in the wrong place?  Where do you expect them to go?

SteveGilvarry, 2015-04-01:
Here is @onlyjob 's intro https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=760314#67.
As a user of zoneminder, I certainly appreciate you taking the task on, the project didn't really have anyone confident enough to take on maintainer for debian, seems to scare people. I tried to play with building a package in debian with cmake and gave up, so if you want to update debian_cmake and then need someone to test and play with it I am happy to help, as far as my skills will allow.

onlyjob, 2015-04-01:
On Tue, 31 Mar 2015 17:07:10 Andrew Bauer wrote:
> Well, what do you expect?  You show up out of nowhere a total stranger and
> state the sky is falling.

Apologies for not bothering to establish my credibility. ;)
I simply believe I don't need to as the problem speaks for itself.
I have no intention to be overly dramatic, just reporting a bug
that's all...

 
> You've done a lot of making claims of broken-ness but you've provided very
> little factual information about how exactly you set up your environment,
> and only now have you even stated what you are [...]

Well I'm building package in clean Debian "unstable" or "testing"
(a.k.a. Jessie) environment using usual packaging tools
(e.g. "pbuilder", "debuild", "dpkg-buildpackage").

> So for the moment, yeah, I am sceptical, because the facts don't indicate
> what you describe. We've been building packages based on the current method
> for about year now without experiencing the [...]

I realise that my report is missing some details. After over 18 hours
of fiddling with this problem I can describe precisely what is wrong:

When Debian package is built `dh_auto_build` first invoke build system
(CMake in this particular case) to stage build (i.e. `make build`)
in in the temporary build directory:

    /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu

During build stage `make --makefile=MakefilePerl pure_install` put
`ZoneMinder.pm` to `scripts/ZoneMinder/output/share/perl5` under build
directory:

~~~~
cd /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder && make --makefile=MakefilePerl pure_install
make[4]: Entering directory '/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder'
Installing /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output/share/perl5/ZoneMinder.pm
~~~~

Then `dh_auto_install` perform `make install` stage and install files into
package stage directory (usually "debian/tmp" for packages producing
multiple binary DEBs or "debian/packagename" if only one DEB it to be
generated). Package staging directory is a mini-root file system where all
files will be exactly as they will be installed. So if file
`/usr/bin/zmupdate.pl` is to be installed it will appear as
`debian/tmp/usr/bin/zmupdate.pl` or as `debian/zoneminder/usr/bin/zmupdate.pl`.

The following is run during "install stage":

~~~~
cd /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder && perl Makefile.PL FIRST_MAKEFILE=MakefilePerl PREFIX="/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output" LIB="/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output//share/perl5" INSTALLSITEMAN3DIR="/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output/share/man/man3"
[...]
cd /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder && make --makefile=MakefilePerl pure_install
make[5]: Entering directory '/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder'
Installing /mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/debian/tmp/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output/share/perl5/ZoneMinder.pm
~~~~

As you can see `ZoneMinder.pm` erroneously saved into

~~~~
debian/tmp/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output/share/perl5/ZoneMinder.pm
~~~~

because CMake invoke `make --makefile=MakefilePerl pure_install` again
with `LIB` containing absolute path which is appended to `PREFIX`
containing absolute path as well.

Then later during "install stage" the following is executed:
[... 46 lines cut ...]
I hope I just answered this above.

> I'm not asking if you like it or not.  That folder is there
> to make the Debian package maintainer's life easier so ultimately you will
> likely get the chance to make it look however you want if you stick around.

I'm motivated to maintain Zoneminder Debian packaging only in one place
 -- in Debian. Of course you're welcome to take and incorporate all our
packaging improvements if you wish.

>   If you build a package with those scripts, what is wrong with resulting
> package?  What files (Perl modules) are in the wrong place?  Where do you
> expect them to go?

Although `distros/debian_cmake` suffers from the same build system disfunction
it incidentally produces .DEB package where Perl modules are installed to the
correct directory. However CMake build system unnecessary create files in

~~~~
debian/tmp/mnt/tmpssd/src/zoneminder/ZoneMinder-1.28.1/obj-x86_64-linux-gnu/scripts/ZoneMinder/output/share/perl5/ZoneMinder.pm
~~~~

during package build which is a problem.

I hope now this situation is more understandable.

knnniggett, 2015-04-01:
Excellent.  Thank you for taking the time to document what you are doing. I normally work with RPM packages rather than Deb so give me a bit to digest this and then come up with a solution.

onlyjob, 2015-04-01:
No worries. Too bad email reply broke formatting. It is not fixable through edit... :(

knnniggett, 2015-04-06:
Note that I'm still working on this.  While I have not pinned down what is causing the problem, I did notice that the makemaker documentation states the PREFIX & LIB parameters are not recommended.  Apparently PREFIX does not behave like one would expect.  Perhaps that has got something to do with it.

I am reworking how zoneminder builds its perl modules to use parameters that are recommended. e.g. INSTALLDIRS & DESTDIR.

onlyjob, 2015-04-06:
Thanks. '.packlist' is not needed; Earlier I added workaround to `debian/rules` to delete this file in order to avoid installing it. Also Perl modules' templates (*.pm.in) were installed along with generated modules so I had to remove *.pm.in as well...

knnniggett, 2015-04-07:
Let me know your feedback for #792.
By design, that PR has a net change of zero for redhat distros.  What I am hoping is that, by getting away from PREFIX & LIB makemaker parameters, the funny business going on in your build will go away.

onlyjob, 2015-04-08:
It is much better -- thank you very much.
I think that Perl-related improvements will be eventually useful for RH distros when they upgrade.

knnniggett, 2015-06-17:
Think this has been resolved.  Gong to close this one out.

closed 2015-06-17 by knnniggett