ZoneMinder/zoneminder #770: JSON::Any is deprecated

onlyjob, 2015-03-31:
`JSON::Any` (provided by `libjson-any-perl` on Debian) contains the following deprecation notice:

> The original need for JSON::Any has been solved (quite some time ago
    actually). If you're producing new code it is recommended to use
    JSON::MaybeXS which will optionally use Cpanel::JSON::XS for speed
    purposes.

>  JSON::Any will continue to be maintained for compatibility with existing
    code, but for new code you should strongly consider using JSON::MaybeXS
    instead.

`JSON::Any` is used in `General.pm`.

knnniggett, 2015-05-11:
JSON::MaybeXS does not appear to be a drop in replacement ... or maybe JSON::Any never worked in the first place.  I get errors in the log if I try to substitute the encode_json method for the former objToJson method.

The code currently checks for JSON::Any with an Eval Block.  If the expected results are not returned, zoneminder does all the encoding/decoding internally w/o calling any module.

I'm tempted to just delete all the checks for JSON::Any.

SteveGilvarry, 2017-03-18:
Review and delete if no drop in replacement.

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

JasminSAB, 2018-06-16:
STOP SPAMING ALREADY

onlyjob, 2018-06-16:
@JasminSAB, I was not spamming and I specifically deleted my comments to avoid noise.

I had to touch this and other tickets to prevent their automatic closing by _stale bot_.

Is there a better way to do that?