ZoneMinder/zoneminder #771: FTBFS on kFreeBSD

onlyjob, 2015-03-31:
Zoneminder 1.28.1 FTBFS on kFreeBSD:

~~~~
[ 22%] Building CXX object src/CMakeFiles/zm.dir/zm_logger.cpp.o
cd /«PKGBUILDDIR»/dbuild/src && /usr/bin/c++    -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2  -Wall -D__STDC_CONSTANT_MACROS -O2 -I/«PKGBUILDDIR»/dbuild/src -I/«PKGBUILDDIR»/src -I/«PKGBUILDDIR»/dbuild    -o CMakeFiles/zm.dir/zm_logger.cpp.o -c /«PKGBUILDDIR»/src/zm_logger.cpp
/«PKGBUILDDIR»/src/zm_logger.cpp: In member function 'void Logger::logPrint(bool, const char*, int, int, const char*, ...)':
/«PKGBUILDDIR»/src/zm_logger.cpp:531:29: error: 'SYS_gettid' was not declared in this scope
         if ( (tid = syscall(SYS_gettid)) < 0 ) // Thread/Process id
                             ^
/usr/bin/cmake -E cmake_progress_report /«PKGBUILDDIR»/dbuild/CMakeFiles 12
src/CMakeFiles/zm.dir/build.make:264: recipe for target 'src/CMakeFiles/zm.dir/zm_logger.cpp.o' failed
make[3]: *** [src/CMakeFiles/zm.dir/zm_logger.cpp.o] Error 1
~~~~

See full build logs:

 * https://buildd.debian.org/status/fetch.php?pkg=zoneminder&arch=kfreebsd-amd64&ver=1.28.1-1&stamp=1427780071
 * https://buildd.debian.org/status/fetch.php?pkg=zoneminder&arch=kfreebsd-i386&ver=1.28.1-1&stamp=1427779905

connortechnology, 2015-03-31:
There is a branch here called freebsd_fixes which contains my work making zm build on freebsd.

connortechnology, 2015-04-01:
I have made some updates to the freebsd_fixes branch I think it will pass travis now which means I might make  PR out of it.

onlyjob, 2015-04-01:
Here is similar bug where possible solutions are discussed: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=725383

onlyjob, 2015-04-03:
I found workaround for this problem in QEMU:

 * http://sources.debian.net/src/qemu/1:2.1%2Bdfsg-11/util/oslib-posix.c/?hl=77#L77

If they are doing it properly then the following patch could be of help:
~~~~
--- a/src/zm_logger.cpp
+++ b/src/zm_logger.cpp
@@ -527,9 +527,13 @@
     #endif
 
         pid_t tid;
 #ifdef HAVE_SYSCALL
+#if defined(__linux__)
         if ( (tid = syscall(SYS_gettid)) < 0 ) // Thread/Process id
+#else
+        if ( (tid = getpid()) < 0 )
+#endif
 #endif // HAVE_SYSCALL
         tid = getpid(); // Process id
 
         char *logPtr = logString;
~~~~

Please review, I'm not a CPP developer so I have no idea if it will work...

onlyjob, 2015-04-18:
Just tried to build latest `master` with new FreeBSD fixes on kFreeBSD. It still FTBFS:

~~~~
/var/tmp/src/zoneminder/master/src/zm_logger.cpp: In member function ‘void Logger::logPrint(bool, const char*, int, int, const char*, ...)’:
/var/tmp/src/zoneminder/master/src/zm_logger.cpp:541:29: error: ‘SYS_gettid’ was not declared in this scope
         if ( (tid = syscall(SYS_gettid)) < 0 ) // Thread/Process id
                             ^
~~~~

knnniggett, 2015-04-18:
This seems to provide some helpful background:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=725383

connortechnology, 2015-04-20:
I guess I'll have to install kfreebsd.

onlyjob, 2015-04-24:
Thanks, @connortechnology. With 209f23c compilation progressed a bit further until it failed as follows:

~~~~
[ 51%] Building CXX object src/CMakeFiles/zm.dir/zm_signal.cpp.o
/var/tmp/src/zoneminder/master/src/zm_signal.cpp: In function ‘void zm_die_handler(int, siginfo_t*, void*)’:
/var/tmp/src/zoneminder/master/src/zm_signal.cpp:75:33: error: ‘struct mcontext_t’ has no member named ‘gregs’
   ip = (void *)(uc->uc_mcontext.gregs[REG_EIP]);
                                 ^
/var/tmp/src/zoneminder/master/src/zm_signal.cpp:75:39: error: ‘REG_EIP’ was not declared in this scope
   ip = (void *)(uc->uc_mcontext.gregs[REG_EIP]);
                                       ^
src/CMakeFiles/zm.dir/build.make:813: recipe for target 'src/CMakeFiles/zm.dir/zm_signal.cpp.o' failed
~~~~

connortechnology, 2015-04-24:
ea795c4 should fix this.

onlyjob, 2015-04-25:
With ea795c4 Zoneminder builds successfully on kFreeBSD. Thanks, Isaac.

onlyjob, 2015-05-08:
Any chance to merge kFreeBSD fixes?

knnniggett, 2015-05-08:
done.

commit 209f23cc (2015-05-08): Fix build issues on kFreeBSD. Fixes #771
commit 5add4422 (2015-05-08): Merge pull request #822 from ZoneMinder/fix_kFreeBSD

Fix build issues on kFreeBSD. Fixes #771
closed 2015-05-08 by knnniggett