ZoneMinder/zoneminder #773: require_once() misplaced in index.php causes a "Call to undefined function" error

luis-castro, 2015-04-01:
Steps to reproduce:
1. Clear all cookies related to you ZoneMinder site
2. Open your ZoneMinder site
3. In IE you'll get an "Internal Server Error". In Firefox you'll get just a blank page
4. An error will appear on Apache log:
[Tue Mar 31 18:06:26.377257 2015] [:error] [pid 11519] [client 192.168.1.12:49346] PHP Fatal error:  Call to undefined function Fatal() in /usr/local/share/zoneminder/www/index.php on line 82

Solution:
Moving line 83 to line 59 (before "if ( isset($_GET['skin']) )") solves the problem.

This might be the patch, but as I am not used to git yet, I dont wan't to make a mess in your sources:

--- index.php   2014-12-29 23:48:19.000000000 -0600
+++ index-fix.php       2015-01-23 12:04:59.499078984 -0600
@@ -56,6 +56,8 @@
 }
 define( "ZM_BASE_URL", $protocol.'://'.$_SERVER['HTTP_HOST'] );

+require_once( 'includes/config.php' );
+
 if ( isset($_GET['skin']) )
     $skin = $_GET['skin'];
 elseif ( isset($_COOKIE['zmSkin']) )
@@ -98,7 +100,6 @@
        setcookie( "zmCSS", $css, time()+3600*24*30*12*10 );
 }

-require_once( 'includes/config.php' );
 require_once( 'includes/logger.php' );

 if ( ZM_OPT_USE_AUTH )

connortechnology, 2015-04-01:
This is actually something I fixed in master while working on other things.

I will fix a fix to a release-1.28.2 branch and maybe we can get another release out soon.

luis-castro, 2015-04-01:
OK, great! Should I close this issue now?

connortechnology, 2015-04-01:
No... let's wait until we actually release 1.28.2

closed 2015-08-10 by knnniggett