ZoneMinder/zoneminder #783: Switching USE_DEEP_STORAGE deleted all events.

onlyjob, 2015-04-04:
On 1.28.1 using web interface I've changed _Options-->Paths-->USE_DEEP_STORAGE_ and restarted Zoneminder. As result all my events were deleted and all contents of `/var/cache/zoneminder/events` were gone quite unexpectedly.

I would expect at least warning in USE_DEEP_STORAGE' description....

To avoid loss of sensitive data please at least inform user about consequences.

knnniggett, 2015-04-04:
It shouldn't delete events.  

I know I didn't lose any events the last time I ran it, but that was a long time ago.  You have to run "zmupdate.pl --migrate-events" to convert your filesystem, but I don't recall if that is supposed to happen automatically or not.  Have to look.  Back in the day I seem to recall running that command myself, from the command line, which produced output indicating what it was doing.

knnniggett, 2015-04-04:
Further inspection hints that this looks to be a documentation issue (or lack thereof).
I was not able to find anything that automates this process, so I will update the documentation accordingly.

Here is the proper way to convert events:
- Stop zoneminder
- If you have the available storage, back up your event filesystem & zm dB
- Enable Deep Storage under options
- Run "zmupdate.pl --migrate-events" from the command line
- Monitor the output for errors
- Start zoneminder after the conversion completes.

knnniggett, 2015-04-04:
Updated with these commits:
3e8c894a3eeac6ab3f93227fdfdb9228d18f9c0c
bde870e

onlyjob, 2015-04-05:
Thanks. Let's remove phrase "If you are upgrading from previous version of Zoneminder" please? I was not upgrading, just operating 1.28.1 (as you may recall I'm still new to Zoneminder). Events was deleted by `zmaudit`, right? May be mention that as well?

onlyjob, 2015-04-05:
There is more to this: `--migrate-events` is an undocumented option of `zmupdate.pl`.
Please update embedded POD documentation.

onlyjob, 2015-04-06:
I'm not sure why but even after updating with 3e8c894 URL `zm/index.php?view=optionhelp&option=ZM_USE_DEEP_STORAGE` still show old help even after restart of all services...

onlyjob, 2015-04-06:
I also feel that we need 

 * either to add confirmation on changing that option to the web interface;

 * or (better) remove this option from web interface and implement it in `zmupdate.pl` instead.

Latter seems to be much safer and easier because once changing database structures is required it make sense to make such change only together with database update.

knnniggett, 2015-04-12:
I'm still thinking about this.
The migrate-events process in zmupdate apparently does turn on DEEP_STORAGE for you:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/zmupdate.pl.in#L322

I also noticed that some of zoneminder's config items have been placed into a category of "hidden":
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/ConfigData.pm.in#L365

I need to test whether or not hiding the variable simply hides it from appearing under options (what we want) or if if removes access to it entirely.

knnniggett, 2015-04-12:
See #802

onlyjob, 2015-05-08:
I think #802 qualifies for new release -- after all it prevents data loss... And we will be able to close this issue...

knnniggett, 2015-08-22:
I am not willing to keep resolved issues open simply because they are not yet part of an official release.  That just makes our long list of issues harder to manage.  Rest assured, this will be part of our next release.

closed 2015-08-22 by knnniggett

onlyjob, 2015-08-23:
Sure, that makes sense. However I was hoping that serious issues like this could provide enough motivation to make a point release in order to deliver fix ASAP...