ZoneMinder/zoneminder #784: please update bundled Mootools

onlyjob, 2015-04-05:
Let's upgrade mootools from v1.3.2 to 1.4.5 please.
1.4.5 is the version that soon to-be-released Debian Jessie is shipping.
Official Debian packaging of Zoneminder uses system (not bundled) version of mootools.

From the brief look it seems that code changes will not be needed but I'd like someone to confirm that. Thanks.

knnniggett, 2015-04-12:
ZoneMinder uses both mootools-core and mootools-more.
mootools-core has a 1.4.5 release, but mootools-more does not.  
Looks like our choices are:
1.4.0.1
1.5.0
1.5.1

If we upgrade mootools-more to 1.5.x then we might as well do the same thing to mootools-core.
This of course, assumes that upgrading doesn't break too much in zoneminder.

Which versions would you prefer to upgrade to?

onlyjob, 2015-04-12:
I see... Both have 1.5.0 or 1.5.1 versions:

 * https://github.com/mootools/mootools-more/releases
 * https://github.com/mootools/mootools-core/releases

knnniggett, 2015-04-12:
Let's try 1.5.1 for core & more.  I'll put something together.

knnniggett, 2015-04-12:
see #803

knnniggett, 2015-06-17:
Mootools has been successfully updated to the most recent versions possible.  Closing.

closed 2015-06-17 by knnniggett