ZoneMinder/zoneminder #785: please update bundled jQuery

onlyjob, 2015-04-05:
Let's upgrade jquery from v1.4.2 to 1.7.2 please.
1.7.2 is the version that soon to-be-released Debian Jessie is shipping.
Official Debian packaging of Zoneminder uses system (not bundled) version of jquery.

I'm not sure if code changes will be needed but new version appears to be backwards compatible. Thanks.

knnniggett, 2015-04-12:
I think I want to let @kylejohnson respond to this.
I see on the jquery site that there are many versions to choose from.  If someone could give me a specific version number, I can create the pr.

onlyjob, 2015-04-12:
As far as I can tell jQuery 1.7.2 works just fine in Zoneminder. In fact I've never used Zoneminder with older bundled jQuery 1.4.2... I'm all for conservative upgrade, not to the latest version but to the well tested 1.7.2 for now.

knnniggett, 2015-04-12:
I noticed the new API Kyle is working on has some references to jquery in it.  I want make sure it doesn't affect that.

kylejohnson, 2015-04-13:
Any reason not to use the latest 1.11.2 version?  
The API should not be affected at all.

onlyjob, 2015-04-15:
It is good to know that jQuery maintains stable API.  
Zoneminder users in Debian rely on system jQuery (and therefore will not benefit from upgrade of a bundled jQuery). At this time Debian ship libjs-jquery/1.7.2 so naturally Zoneminder will be used with this particular version. I suggest upgrading to this particular version because it is the most tested one.

Please remember to create un-versioned symlink `jquery.js` --> `jquery-1.7.2.js` and update reference in `skins/classic/includes/export_functions.php` -- this will simplify future upgrades.

Also please never ever commit minified javascripts. They are same as pre-compiled binaries -- unreadable, unmodifiable, and non-distributable in Debian.

onlyjob, 2015-04-15:
See #809

onlyjob, 2015-04-16:
All good (merged), closing...

closed 2015-04-16 by onlyjob

onlyjob, 2015-04-24:
@kylejohnson FYI [JQuery web site](https://jquery.com/download/) says:

> The jQuery 1.x line had major changes as of jQuery 1.9.0. We strongly recommend that you also use the jQuery Migrate plugin if you are upgrading from pre-1.9 versions of jQuery or need to use plugins that haven't yet been updated. Read the jQuery 1.9 Upgrade Guide and the jQuery 1.9 release blog post for more information.

I reckon we'll need more caution to upgrade to jQuery v1.9.0 and above...

commit 91832e57 (2015-04-15): Merge pull request #809 from onlyjob/jquery

upgrade bundled jQuery (Closes: #785)
commit b2f6bfb5 (2015-04-15): upgrade bundled jQuery (Closes: #785)