ZoneMinder/zoneminder #786: ZM_CONTENTDIR issues

onlyjob, 2015-04-05:
As far as I understand `ZM_CONTENTDIR` is meant to point to content directory where events and images are stored. However there are some issues with that:

 * ZM_CONTENTDIR known only to CMake.

 * ZM_CONTENTDIR is used merely to build `zmlinkcontent.sh` which shouldn't be needed at all.

 * symlinks in `images` and `events` from `/usr/share/zoneminder/www` to ZM_CONTENTDIR (i.e. `/var/lib/zoneminder`) should not be needed if ZM_CONTENTDIR were used as prefix for 
`ZM_DIR_EVENTS` and `ZM_DIR_IMAGES`. This would obsolete `zmlinkcontent.sh` and simplify installation instructions.

onlyjob, 2015-04-05:
Symlinks should not be needed under web directory `/usr/share/zoneminder/www` because web server do not need to export those directories for correct functioning of Zoneminder.
`images`, `events` and `temp` are used only by daemon and/or server-side PHP. The latter should be able to access 'em directly in `/var/lib/zoneminder` without symlinking to `/usr/share/zoneminder/www`...

knnniggett, 2015-04-05:
Cmake is the newer build method for zoneminder with the ultimate goal of deprecating autotools.  The two build methods are not going to have the exact same feature set.  Cmake should have more/better features than autotools, however, since that is what we want to move to long term.

zmlinkcontent.sh is most helpful for those building from source, and the reason we support that is historical (previous discussion, no need to rehash this).  You certainly don't need to call this to set up the symlinks.  Rather you can do it inside your packaging scripts.  Isn't that was is being done here? https://github.com/ZoneMinder/ZoneMinder/blob/master/distros/debian_cmake/preinst

For Fedora & Redhat, I wrote a CMakeLists.txt file that lives under the appropriate /distros subfolder which handles the symlinking, among other things.  However, I get the impression that you like to keep things directly under your own control so I won't be surprised if you prefer not to do it this way (not a personal attack, just an observation).

As far as needing symlinks at all, FHS tells us that application data should live under /var. There seems to be disagreement between redhat & debian whether the files should live under /var/lib or /var/cache, but I think we are splitting hairs for this particular argument.  However, the web user account needs to access these files, so we have to symlink them to be under the webroot a.k.a. /usr/share/zoneminder.

Maybe there is a way to re-write the core of zoneminder so that the data under /var can be accessed without symlinks, but that is not something we are willing to do with the current version of zoneminder. As you know, there are a lot things in zoneminder, cob webs so to speak, that have been the way they are since before any of us starting working with this project.  This is one of those.

@kylejohnson is working on a redesign of the UI, and I think the necessity of symblinks should be addressed in that.

onlyjob, 2015-04-05:
> zmlinkcontent.sh is most helpful for those building from source, and the reason we support that is historical (previous discussion, no need to rehash this). You certainly don't need to call this to set up the symlinks.

Of course. I understand that. My point is that symlinks should not be needed as long as daemons/cgi scripts can access `/var/lib/zoneminder`. The problem is that Zoneminder can't work properly without those symlinks in `/usr/share/zoneminder/www`...

> Rather you can do it inside your packaging scripts. Isn't that was is being done here? https://github.com/ZoneMinder/ZoneMinder/blob/master/distros/debian_cmake/preinst

Not quite. `preinst` merely checks validity of symlinks and aborts on unexpected changes.
A rather unusual thing to do in `preinst`...

Handling symlinks from inside packaging is easy, that's not a problem. It is strange to observe that `events`, `images` and `temp` in `/usr/share/zoneminder/www` are not accessible through web interface yet they are used by Zoneminder daemon and PHP scripts...

> However, I get the impression that you like to keep things directly under your own control so I won't be surprised if you prefer not to do it this way (not an attack on your methods, just an observation).

I'm not sure how did I create that incorrect impression... It's not about control. My goal is to take Debian packaging off your shoulders. When official package is well maintained there is no reason for you to spend time on packaging upstream. I believe your time is better spent on core Zoneminder functionality rather than packaging. My goal is to take care of packaging in Debian. I happen to have pretty good understanding of Debian packaging so I merely use my skills to do the job the best I know.

> As far as needing symlinks at all, FHS tells us that application data should live under /var. There seems to be disagreement between redhat & debian whether the files should live under /var/lib or /var/cache,

Not quite: in Debian many packages keep their data in `/var/lib`. `/var/cache` also can be used for data although it is more suitable for cache. ;) Zoneminder uses `/var/cache` for historical reasons but that's not a problem. It is not against the policy and Zoneminder is certainly not the only package to keep data in `/var/cache`.

> but I think we are splitting hairs for this particular argument. However, the web user account needs to access these files, so we have to symlink them to be under the webroot a.k.a. /usr/share/zoneminder.

Exactly.

> Maybe there is a way to re-write the core of zoneminder so that the data under /var can be accessed without symlinks, but that is not something we are willing to do with the current version of zoneminder.

Fair enough.

> As you know, there are a lot things in zoneminder, cob webs so to speak, that have been the way they are since before any of us starting working with this project. This is one of those.

Of course. Sorry for troubling you with pedantic bug. There are many of greater importance.

@kylejohnson is working on a redesign of the UI, and I think the necessity of symlinks should be addressed in that.

Nice. Thanks.

mastertheknife, 2015-04-11:
ZM_CONTENTDIR was created to assist in installing ZM by new users.
I felt its not the job of cmake to deal with symlinks, so i needed something that does that, does it well and is easy to use.

Why symlinks are needed in the first place? As ` knnniggett ` explained in depth, this is because of the way ZM works right now. The web server needs access to those directories, and you obviously can't place the events folder in /usr/share, it just doesn't belong there. The natural home for this kind of stuff (dynamic content) is /var/lib, so /var/lib/zoneminder makes perfect sense.

closed 2015-05-10 by knnniggett