ZoneMinder/zoneminder #787: man pages please; improve --help option

onlyjob, 2015-04-05:
All binaries installed to `/usr/bin` should have corresponding man pages.
At the moment there are none. Also in most cases `--help` is incomplete as it doesn't even say what the executable is for.
For Perl scripts we can generate man pages from embedded POD documentation.
Binaries need to have their own separate man pages.

knnniggett, 2015-06-20:
I took me a bit to realize why many of ZoneMinder's perl modules have man pages autogenerated for them, but the same thing is not happening for the Perl scripts under `/usr/bin` do not.  The scripts under `/usr/bin` are not being installed via MakeMaker.  I'm investigating the best way to resolve this.

knnniggett, 2015-06-20:
See pr #896 for initial progress on this issue

knnniggett, 2015-06-20:
#896 has been completed.  Please review.

knnniggett, 2015-07-24:
This has been completed. Closing.

closed 2015-07-24 by knnniggett