ZoneMinder/zoneminder #788: zmsystemctl.pl incorrectly installed to '/usr/bin'

onlyjob, 2015-04-05:
`zmsystemctl.pl` is intended to be used internally by zoneminder therefore (unless it is usable by the user) it should be installed into private directory in `/usr/share/zoneminder` (or to `/usr/lib/zoneminder` for binary architecture-dependent executables).

knnniggett, 2015-05-08:
Like all of the applications in the zoneminder suite, zmsystemctl.pl can be called by the end user.  For what purpose?  I'm not sure, yet.  zmsystemctl.pl is rather new.  If the end user needed a way to pro-grammatically restart zoneminder as the web account user w/o providing credentials, then they could use zmsystemctl.pl.

This isn't the only zoneminder application like this.  zmdc.pl is intended to only be used internally, but a highly skilled user could conceivably call it directly.  Then there are all the applications written in C: zma, zmc, zma, etc.  These tools are intended to be called internally, but they can be called directly by the end user. There are times when this can help with troubleshooting.  Another example occurs when the end user wishes to get zoneminder to interact with a third party application or trigger. Sometimes it is necessary to call one or more of these application directly.

I see what you mean, however.  Most distro's have packaging standards that tell the package maintainer to place these applications elsewhere.  However, since zoneminder has been historically written to allow the end user to manually execute any of the varies applications in the zoneminder suite, I am not willing to change the installation folder at this time.

If the wording in the header description in zmsystemctl.pl is hanging you up, then I can remove the phrase "This script is intended to be called internally" if that suites you.

onlyjob, 2015-05-08:
Thanks for explaining. To me personally this issue is more about having man pages for all exposed utilities (i.e. those in `/usr/bin`). If there is no man page then it is not intended for direct use so why not keep it in private directories? I recommend to add man page for all utilities or move 'em to private location. Besides, users still will be able to use those scripts even from `/usr/lib/zoneminder` or from `/usr/lib/zoneminder/bin`. We already have cgi-scripts not in `/usr/bin` and there is no drama about it...

knnniggett, 2015-06-20:
Just realized that this has turned into a duplicate of #787. Closing this one.

closed 2015-06-20 by knnniggett

onlyjob, 2015-06-20:
Not quite a duplicate because unlike other .pl apps, `zmsystemctl.pl` have no POD documentation revealing what is it for and how it works (which is especially bad for binaries exposed system-wide). #787 is more generic case of having no man pages that potentially could be generated from POD markup. I was working on #787 and made some progress improving POD except for undocumented utilities...
I have no objection to closing this bug if you consider lack of any documentation in `zmsystemctl.pl` to be in scope of #787...

knnniggett, 2015-06-20:
Yes, I'll fix that in #787.  I'll also do the same for the C binaries that don't have man pages.

onlyjob, 2015-06-20:
Thank you. Man pages for C binaries will be much appreciated. :)