ZoneMinder/zoneminder #790: DB is used to store essential config paths

onlyjob, 2015-04-06:
Normally variables like `ZM_PATH_SOCKS` should not be stored in the database because they may need to be used by service startup scripts (init.d, systemd).
It would be highly undesirable to write service startup scripts with functionality to fetch configuration from database.
From packaging prospective it is very inconvenient that location of sock files or `PATH_SWAP` can't be changed by re-deploying binary package built with customised `ZM_SOCKDIR` because those locations can be changed only by modifying database. 

Usually path to socks directory is saved to configuration file under `/etc`.
It is unnecessary (and perhaps undesirable) to allow changing such essential variables through web interface. Also there is a considerable security risk when unprivileged user (not root) can change essential service startup parameters.

Besides it is very counter-intuitive that variables like `ZM_SOCKDIR` given on build time are used merely to generate database initialisation script.

knnniggett, 2015-04-12:
Yes, this actually can and does cause problems. 

For example, a user has a functioning zoneminder system on CentOS and then migrates the database to a system running OpenSuse.  Because the packaging standards are different between the two distros, the zoneminder system running on OpenSuse will have been built with different sock, tmpdir, swap, etc folders.  However, the database, having come from a CentOS environment, contains values that don't match the current system.  Thus the OpenSuse zoneminder system becomes broken right out of the box. These kinds of problems are very hard to catch by a new user, and they are very hard to troubleshoot by us because the new user often does not know to tell us that, "Oh, by the way I just migrated this dB from a different environment".

I think it was @PX03AFK that helped identify this was an issue.  What I have been planning to do is work with @kylejohnson to implement an install wizard in the new UI.  One of the things the wizard could do is compare the values in the database to the values zoneminder was built with (they are in ConfigData.pm).  If any values determined to be critical do not match, alert the user and let him decide what to do about it.

I think the solution you have proposed takes a simpler approach, which I like. However, if we go this route, I want to take a look at all of the variables in the paths tab.  Possibly remove the tab altogether.
  
- We just hid DEEP_STORAGE in a different PR
- The DIR_EVENTS, DIR_IMAGES, and DIR_SOUNDS variables are all relative to the web root, and changing them here is not helpful.  Many people don't realize this and have changed DIR_EVENTS to an absolute path pointing to another mount point.  Not only doesn't that work, but that would trigger a mass event deletion much like the issue with DEEP_STORAGE
- All the remaining variables have the same issue as ZM_PATH_SOCKS. e.g. non-portable settings should go in a config file rather than the dB.

I need to take some time to explore this further. Comments are welcome

onlyjob, 2015-04-12:
Thanks. I've already run into those issues even without changing distro or moving database.
Since first install I simply had socks in volatile `/tmp/zm` path which disappeared on reboot and was not re-created... By the way not-verbose-enough #789 errors made it surprisingly difficult to recover...

Perhaps we need to identify variables like `ZM_PATH_SOCKS` and make changes to fetch 'em from config file instead of fetching them from database. For users transition won't be too painful if we document changes in `ChangeLog`. This is unrelated issue but I'd like to see summary of changes in `ChangeLog` which shouldn't be that hard to maintain (it can even be in a form of `git log`)...

davidsainty, 2015-04-13:
Me too.

It's a pain in a package-based system.  For example, in Pkgsrc the ffmpeg dependency has changed from ffmpeg010 to ffmpeg2.  But the path to ffmpeg is in the database, so it's a manual operation to fix the configuration after upgrading the package.

A more obscure problem is if you have more than one full set of packages installed - for example I have three, "current", "rollback" and "next", under /pkg/pkgroot{1,2,3} - I rotate through which one is the current default build of packages.  But ZoneMinder is probably the one and only package that can't keep up - the DB paths may well point to my "rollback" set of packages after a system upgrade,

knnniggett, 2015-04-13:
A couple of side notes
@onlyjob The tmpfiles.d service should be used to ensure volatile folders are created properly and always have the correct permissions.  

The template for that is here: 
https://github.com/ZoneMinder/ZoneMinder/blob/master/misc/zoneminder-tmpfiles.conf.in.  

Set up your package to install that config file into /etc/tmpfiles.d.  

Since most modern distros seem to be using private tmp folders, be careful to avoid placing files into these.  For example, if you place the sockfile into a folder that is part of a private tmpfs, then you'll get a situation where the web UI states zoneminder is stopped but systemctl says it is running.

@davidsainty You will need to be more specific with regards to ffmpeg.  Zoneminder stores the path to the ffmpeg binary in the database, but that is only for exporting events.  It doesn't use the ffmpeg binary to stream from IP cameras; it uses the API functions calls for that.  If the path to the ffmpeg binary is incorrect, it may even go unnoticed.  

Are you saying that the path to your ffmpeg binary changes with each upgrade?  Maybe that is a freebsd thing.  

In any case, there is a tool you can use to change most any zoneminder default value, prior to building.  Call it like so before you call cmake/configure:
/utils/zmeditconfigdata.sh ZM_PATH_FFMPEG /path/to/ffmpeg/binary

I'm not familiar with the build environment you have described so I don't know if this will help or not.

onlyjob, 2015-04-13:
Thanks for the hint.  But "The tmpfiles.d mechanism is currently only used by systemd."
With init.d volatile folders are usually handled by the startup script(s).
Such problem existed in the old defunctional packaging.
"tmpfiles.d" is relatively new thing and latest Zoneminder release doesn't even have this template yet. I'll probably add tmpfile to Debian packaging. Thanks.

davidsainty, 2015-04-13:
The ffmpeg binary will sometimes change location, yes.  Actually, this is on Ubuntu Linux, but I use NetBSD's Pkgsrc for the software management.

To give a more concrete example, I may have two installations of ZoneMinder - based under /pkg/pkgrooot1 and /pkg/pkgroot2.  The "live" one may be /pkg/pkgrooot1, and the "upgrade" may be /pkg/pkgroot2.  The way I manage this is to have a symlink /pkg/live -> /pkg/pkgrooot1.

There is only one set of database data files though.

Now lets say that I want to test the upgrade.  I may shut down ZoneMinder and try running it from /pkg/pkgroot2 explicitly.  Perhaps via: /pkg/pkgroot2/bin/zmpkg.pl start

In an ideal world, starting the system from pkgroot2 would use "/pkg/pkgroot2/bin/ffmpeg2" (because it's a brand new ZoneMinder version 1.28.1).

In same ideal world, starting the system from pkgroot1 would use "/pkg/pkgroot1/bin/ffmpeg010" (because it's an old ZoneMinder version 1.25.0).

Depending on what I'm doing - running the live version, testing a new version, switching the live symlink to the new version base, or rolling back to the old version after finding problems - the correct paths should be determined by which code is currently running, not the single DB that's shared between all of them.

So I think automatically editing the database is not right, because it all depends on which path you chose to run at run time.  For clarity, here's an example zm.conf (from pkgroot2):

% cat /pkg/pkgroot2/etc/zm.conf
# ==========================================================================
#
# ZoneMinder Base Configuration, $Date$, $Revision$
#
# ==========================================================================
#
# This file is generated by 'configure'. Care should be taken if manually
# editing this file as an changes may be overwritten by subsequent configuration
# or installations.
#

# Path to installed data directory, used mostly for finding DB upgrade scripts
ZM_PATH_DATA=/pkg/pkgroot2/share/zoneminder

# Path to ZoneMinder binaries
ZM_PATH_BIN=/pkg/pkgroot2/bin

# Path to ZoneMinder libraries (none at present, for future use)
ZM_PATH_LIB=/pkg/pkgroot2/lib

# Path to ZoneMinder configuration (this file only at present)
ZM_PATH_CONF=/pkg/pkgroot2/etc

# Path to ZoneMinder web files
ZM_PATH_WEB=/pkg/pkgroot2/share/zoneminder/htdocs

# Path to ZoneMinder cgi files
ZM_PATH_CGI=/pkg/pkgroot2/libexec/zoneminder/cgi-bin

# Username and group that web daemon (httpd/apache) runs as
ZM_WEB_USER=www
ZM_WEB_GROUP=www

# ZoneMinder database type: so far only mysql is supported
ZM_DB_TYPE=mysql

# ZoneMinder database hostname or ip address
ZM_DB_HOST=localhost

# ZoneMinder database name
ZM_DB_NAME=zm

davidsainty, 2015-04-13:
Ugh, sorry for the broken markup - and for some reason I can't edit the comment either.

In terms of the use of the binary - yes, I'm aware it's not totally essential.  I raise this data point mainly to demonstrate a technical reason for shifting the path from the DB.  In reality I configure the DB path to /pkg/live/bin/ffmpeg2 - which works acceptably well for version management, and generally can be left alone so long as the ffmpeg major version hasn't changed.

PR #1908 (2017-06-13): Migrate values from Options -> Paths to zm.conf
fixes #790

The values on the paths tab are not portable. When the end user backs up the database from one system and restores it to another, the Options -> Paths values must be reviewed for accuracy on the new system. Much time can be spent troubleshooting if the end user does not know to check this. This pr does away with that issue by migrating the values out of the database and into zm.conf.

This pr also simplifies the next pr I plan to make, which will remove zoneminder's reliance on symlinks.

This needs testing with debian packages built  [... cut]
closed 2017-06-13 by connortechnology