ZoneMinder/zoneminder #797: How do I add authentication to ZM APIs?

arjunroychowdhury, 2015-04-10:
Hello, on recommendation from knniggett I just installed ZM 1.28.1 from source via cmake. This installs the APIs too.

I am refactoring my mobile app to use the APIs. How do I go about adding authentication to the APIs? As of today, I can do http://myserver/zm/api/monitors.json from any browser and it prints out all my details including passwords. Any tips on how to authenticate this api layer?

arjunroychowdhury, 2015-04-10:
Specifically, after installing via cmake, the APIs were not configured for us.
I add RewriteBase /zm/api to  www/api/.htaccess and www/api/app/webroot/.htaccess to make the APIs load

knnniggett, 2015-04-11:
@kylejohnson is the one to answer questions about the API, but I think he is still working on a bathroom remodel.  I saw a picture of it the other day and it looked like he had a ways to go.  So you are not being ignored. :-)

arjunroychowdhury, 2015-04-11:
A man needs his bathroom! I'll wait patiently. I just hope he doesn't tell me to install another branch. I went through hell making cmake work. Well, if kyle helps me on this, maybe I can help him finish his bathroom! It looks like we are both in Maryland :-D

kylejohnson, 2015-04-11:
Hi there - I've just returned from vacation in Colorado!  The bathroom is coming along (slowly).

I regret to report that the current API does **not** support authentication.  I did not choose to add this feature into the first release of the API as there were no known clients using it (including our own), and I was interested in porting functionality. 

I'm going to move the rest of this convo. over to #799

closed 2015-04-11 by kylejohnson

mastertheknife, 2015-04-11:
Most REST APIs do not deal with authentication but rather let the web server take care of it. This is because REST is stateless.
You can probably use apache's built in authentication for this (.htaccess and .htpasswd).
See here: http://httpd.apache.org/docs/2.2/mod/mod_auth_basic.html

arjunroychowdhury, 2015-04-11:
@mastertheknife, first thanks for this tip.I am going to set it up that way so at least my APIs are not open for all.

If I may put forward some more thoughts:
a) I'm not sure if authentication makes the REST APIs stateful. The auth should turn into a per API token rendering it completely stateless, is it not?
b) As a consumer of the APIs, when developing my app, I really don't want to ask the application user any more than one username and one password. With that one auth access, I should be able to perform all the API operations, as well as view live feed and event recordings. From what I understand today, if @kylejohnson is planning to keep API auth separate from ZM auth (I guess the same applies to your suggestion here too), then one would need to make sure that the camera image feeds are also available via the API layer.