ZoneMinder/zoneminder #805: Input/Output error is fatal to zma

connortechnology, 2015-04-13:
If for some reason zma can't create mkdir the directory for the event it is fatal.  This results in the image getting lost.  

What should happen?  Well, long term I hope to have support for multiple storage areas, so perhaps it could try somewhere else.  

Short term, perhaps it should try again.  Maybe it should dump to /tmp.  

I think the #1 priority in a video surveillance solution is getting the image to disk.

Thoughts?

knnniggett, 2015-04-13:
I think this is one of those things that is trickier than it sounds.

Trying again sounds reasonable, but we would need to not overwhelm the system with re-tries.

We would have to be really careful if we stored events anywhere other than the location(s) the user has configured for storage.  Writing to /tmp usually means writing to memory, which could fill up quickly, but writing to a real folder that sits on the filesystem, such as /var/tmp, risks filling up a partition the system needs to function correctly.  

If the problem persists, what would we do if the temporary storage fills up?  
I guess at some point we would have to lose events.
Once the write problem is resolved, should zoneminder automatically move the events back to the original intended location?

connortechnology, 2015-04-13:
Well in my case today it was an nfs timeout that caused the failure.  A second attempt would have succeeded.  We can't wait too long or else we won't be doing analysis.  I think perhaps 2 or at most 3 attempts, then give up and try the backup storage area.

The backup storage area would have to be configurable. The images could be moved into proper position later by zmaudit.  

Ideally at this point someone would notice the errors and fix the underlying problem.

davidsainty, 2015-04-13:
It's a bit of a niche problem and solution.  Wouldn't it be better to
require and use local storage for all in-flight events, but support
automated and transparent archival storage to NFS (or some other slow or
remote service)?

Normally if you have an application that requires rock solid storage, you
use rock solid storage.  E.g. even database servers don't support fail-over
to some local short term storage on errors.

connortechnology, 2015-04-14:
There is no such thing as rock solid storage IME. SAN's have network problems, local storage has failing hard drives, even my 64TB raid6 with two spares is glitchy.  (Never buy Promise hardware).  

 I think the multiple storage areas thing is the best way to go.  Can't write to the usual spot, write to some other spot. I think that's how I will proceed.

davidsainty, 2015-04-17:
Sure, everything can break - including rocks :)  But local storage is
usually solid enough, because if it dies probably the OS is about to die
too.  Fail-over becomes a moot point.

To do storage failover right sounds like a massive amount of effort (and
managing incredibly complicated failure cases - which might be
counter-productive in terms of system reliability) for a very small number
of users - who could solve the problem just by storing the data in a more
reliable destination in the first place.

Don't get me wrong, I quite like the idea of multiple storage areas in
principle.  I guess I'm just advocating Keeping It Simple.

But if you do do it, can you extend it to do automatic mirroring too?  I.e.
unconditionally writing to multiple destinations rather than just writing
on failure. Possibly via plugins like "scp to an off-site server".  That
would be a boon to a security system.

connortechnology, 2015-04-17:
My main reason for doing multiple storage areas is the multi-server case.  Every remote storage type introduces way too much latency.  Each server needs local storage, but needs to be able to access each other's remote storage.  I have a client that desperately needs this, so it is going to happen.

The added benefit is the case where your local storage has a problem, you can use whatever other storage is available.

The automatic mirroring is an interesting idea.  I will certainly do that.  I don't know about scp'ing... I can't help but feel that more complicated stuff should happen as a separate background task.  But I can design with it in mind so that it can be added easily in future.

knnniggett, 2015-05-12:
At my dayjob, when we have a surveillance project requiring networked storage, we carve out a dedicated vlan between the server(s) and the storage, with jumbo frames enabled.  Connections are iscsi.  If the project has a SAN, we will map all the extra nic's on the recorder via multipath to the SAN, but even if we've only got a NAS, a single gigabit link has always worked just fine.  

We have never had any issues with latency. For us, this configuration has been rock solid.

Of course, we aren't using ZoneMinder, but what I'm getting at is the issue you are experiencing might have something to do with the way the multi-server stuff works.  Once you get the multi-server merged into master, then we can put it to the test and perhaps someone might figure out a way to improve on it.

connortechnology, 2015-05-12:
well.. I think the issue becomes with 80 cameras over 5 servers things like running the filters and audit traverse too much of the directory tree and with all those files, it is slow and the traversal kills the caches.

The h264 video branch will help alot, and part of the multi-server code will include being able to run the filters on specific hosts (right now it runs on all).

closed 2018-06-22 by stale[bot]