ZoneMinder/zoneminder #806: PHP session locks serialize access to all scripts causing UI hangs

dvsingletary, 2015-04-14:
In index.php, a PHP session is started/resumed with a call to session_start(), however the session file is never closed explicitly with session_write_close(), so this effectively serializes all requests to the zoneminder UI.  This means if a request consumes any time (exporting of video file, requesting a stale video stream, etc) then it will block all other web UI requests until it is complete.

I noticed this mostly when I was watching a live feed, and then updated the source settings for that camera.  When this is done, the capture process is restarted causing the socket that transfers frames to the stream server to go away.  In ajax/stream.php if the socket file is missing it calls sleep(1) up to 10 times waiting for the file to appear on the assumption that the request beat the creating of the socket.  Overlapping calls to this effective hang the UI completely until the browser window with the stale live feed is closed.

To fix the above-mentioned issue I added "session_write_close()" in stream.php above the while loop that sleeps.  This is just a local hack... the correct way to do this is to not leave the session file open if anything you are doing consumes time.  It's best just to open it, get what you need, and close it.  You can always reopen it later in the script (session_start()) if you need to save updated state.

connortechnology, 2015-04-15:
OMG thank you so much figuring this out!  It has been annoying me forever! Now that I know where the problem is I'm sure I can help with perfecting a fix.

dvsingletary, 2015-04-15:
so... I've already altered my local hack for this.  Basically, at the end of the index.php script there is a condition to either include something from the ajax path, or something from the include path.  The time-consuming functions seem to be mostly ajax-related, so I just call session_write_close() before including the ajax file.  There is one ajax file that needs to write session variables, event.php, and I session_start() in here before I modify them so that they get written out properly.

connortechnology, 2015-04-20:
I've gone farther with my PR.  I close the session regardless of whether it is ajax or a view.  I feel that if a view needs the session it can re-open it.

dvsingletary, 2015-04-20:
Yup. That's a great model. In general never leave the session file open longer than necessary.

knnniggett, 2015-05-10:
Fixed in #823

closed 2015-05-10 by knnniggett