ZoneMinder/zoneminder #810: bundled CakePHP overhead

onlyjob, 2015-04-15:
It looks like it is unnecessary to install full CakePHP sources.
I'm not sure but probably we can safely drop `web/api/app` and the whole `web/api/lib/Cake/Test` (the latter is over 5 MiB and (as I suspect) it is not used at all).

Which Zoneminder functionality needs CakePHP?

knnniggett, 2015-04-15:
@kylejohnson is developing the new api using cakephp.  Think he is also developing the new ui in cakephp if I recall.  In any case, moving forward we want all third party developers to use the API rather than using something like the xml skin, which is eventually going to be depreciated.

SteveGilvarry, 2015-04-16:
Web/api/app I believe is where the API lives, web/api/lib is the CakePHP framework itself, and I know what you mean about the test folder. But I think trimming it out is removing part of the framework, and same folders still exist in slightly different location in CakePHP 3.0.1. Maybe close this issue in preference to the upgrade one.

kylejohnson, 2015-04-16:
As @knnniggett said, the API uses CakePHP.  The new UI uses the API for data, but is actually written in AngularJS.

#744 will address some of the issues in this ticket, however I don't want to make any large-scale cakephp changes until the new UI and API are in a stable state for general use.  I might upgrade before then, if I run into a bug that blocks me, but I haven't yet.

closed 2015-04-16 by kylejohnson