ZoneMinder/zoneminder #813: left object detection

dvsingletary, 2015-04-16:
I haven't looked at the code in zma yet, but I suspect it wouldn't be too difficult to add a new zone type called "ObjDect", whereby at the conclusion of any event would run a single frame analysis using the first frame of the event as the reference, and the last frame of the event as the frame to analyze.  The object here would be to detect objects left as a result of an event.  The immediate use case for me would be when someone leaves a package on my doorstep -- I don't want an email for every single doorstep motion alarm, but if something was left on my doorstep I definitely want an email =)  If the zone detected a left package, it would just include that zone name in the list of fired zones for that event and you could filter on for email or text notification, etc.

knnniggett, 2015-04-16:
I think @nextime wrote an analytics package for zoneminder. Not sure if he is actively working on zoneminder.  @kylejohnson may know more about this.
I know we talked about adding his work in as a new feature for a future, major release.

knnniggett, 2015-08-09:
Closing due to inactivity. If you write a pr to implement this, feel free to reopen this feature request.

closed 2015-08-09 by knnniggett