ZoneMinder/zoneminder #814: Upgrade bundled CakePHP

onlyjob, 2015-04-16:
Recently CakePHP released new stable version _3.0.1_ while our bundled CakePHP is _2.5.0-RC1_. Let's upgrade please? See https://github.com/cakephp/cakephp/releases

kylejohnson, 2015-04-16:
Nope.  3.x is not backwards compatible.  We won't be upgrading for a loooooong time, at least until the new ui + api is finished.

kylejohnson, 2015-04-16:
That said, we may be able to upgrade to a later 2.x release.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

onlyjob, 2015-04-16:
Fair enough. Then perhaps 2.6.3 would be the good version to upgrade to and it is already in Debian. ;)

SteveGilvarry, 2017-03-18:
Don't want to say it but hoping this is the last version of CakePHP we have.

closed 2017-03-18 by SteveGilvarry