ZoneMinder/zoneminder #817: conflicting license statements

onlyjob, 2015-04-16:
I've noticed that some files have conflicting license statements. For instance the header of file [scripts/ZoneMinder/lib/ZoneMinder/Trigger/Channel.pm](../blob/master/scripts/ZoneMinder/lib/ZoneMinder/Trigger/Channel.pm) claims GPL-2+ but in the end of the same file I found [the following](../blob/master/scripts/ZoneMinder/lib/ZoneMinder/Trigger/Channel.pm#L161) statement:

> This library is free software; you can redistribute it and/or modify
it under the same terms as Perl itself, either Perl version 5.8.3 or,
at your option, any later version of Perl 5 you may have available.

Please clarify whether such `.pm` files are double-licensed or if only one particular license applies.

All `.pm` files in `scripts/ZoneMinder/lib/ZoneMinder/Trigger` are affected, some in `scripts/ZoneMinder/lib/ZoneMinder/Control` as well as all `*.pm` and `*.pm.in` in `scripts/ZoneMinder/lib/ZoneMinder`.

knnniggett, 2015-05-10:
@kylejohnson can you get a hold of Phil and see what he thinks about this one?

web2wire, 2015-05-10:
I'm happy to put everything on the same licence, probably GPL3 nowadays. I suspect the perl ones just had whatever I found in similar modules from elsewhere.

onlyjob, 2015-05-10:
So at the moment some Perl components (modules and scripts where Perl license is mentioned) are dual-licensed but we can upgrade to GPL-3+ (version 3 or above), right?

Since everything is already under GPL-2+ I suppose it already includes GPL-3+...

If everyone agrees we can just drop Perl license statement or (in case we choose to preserve dual-licensing) move them towards the beginning of files...

SteveGilvarry, 2017-03-18:
Review and fix all licenses

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

996b, 2019-10-19:
Yeah not really knowing much about this trying to learn something