ZoneMinder/zoneminder #819: Empty string if missing index in lang files

manupap1, 2015-04-18:
No string is displayed in web pages if the requested index is missing in SLANG array.
I think the index should be displayed as a fallback method.
This simple function could be used in `web/includes/lang.php`:
```php
function translate( $name )
{
    global $SLANG;
    if ( array_key_exists( $name, $SLANG ) )
        return $SLANG[$name];
    else
        return $name;
}
```
This script could update all files in a row: 
```bash
#!/bin/bash
FPATH="web/ajax web/skins"
for folder in $FPATH
do
    find $folder -type f -name "*.php" -exec sed -i "s/\$SLANG\['\([a-zA-Z0-9\]*\)']/translate\('\1'\)/g" {} \;
done
```

knnniggett, 2015-05-10:
Please test #833

SteveGilvarry, 2015-06-16:
Closed via #833 and #845

closed 2015-06-16 by SteveGilvarry