ZoneMinder/zoneminder #827: curl using HEAD instead of GET for http requests results in 404 on some servers

jakobmetzger, 2015-04-28:
curl by default uses HEAD to retrieve info which gives a 404 error on some servers. More info here https://stackoverflow.com/questions/10404727/curl-php-http-code-says-404-but-browser-says-its-200

I discovered this while trying out IPCam(http://skjm.com/ipcam/) as a server on an iPhone. The image loads normally in a browser but inside of zoneminder just results in an all blue image. 

```
# curl -I --verbose 10.0.1.71/image.jpg
* About to connect() to 10.0.1.71 port 80 (#0)
*   Trying 10.0.1.71...
* Adding handle: conn: 0xea1860
* Adding handle: send: 0
* Adding handle: recv: 0
* Curl_addHandleToPipeline: length: 1
* - Conn 0 (0xea1860) send_pipe: 1, recv_pipe: 0
* Connected to 10.0.1.71 (10.0.1.71) port 80 (#0)
> HEAD /image.jpg HTTP/1.1
> User-Agent: curl/7.32.0
> Host: 10.0.1.71
> Accept: */*
> 
< HTTP/1.1 404 Not Found
HTTP/1.1 404 Not Found
< Content-Length: 38
Content-Length: 38

< 
* Excess found in a non pipelined read: excess = 38 url = /image.jpg (zero-length body)
* Connection #0 to host 10.0.1.71 left intact
=========================================================
```
Showing the difference by manually requesting through telnet
```
# telnet 10.0.1.71 80
Trying 10.0.1.71...
Connected to 10.0.1.71.
Escape character is '^]'.
GET /image.jpg HTTP/1.1

HTTP/1.1 200 OK
Cache-Control: no-cache, no-store, must-revalidate
Pragma: no-cache
Content-Length: 18253
```

Showing how HEAD with the same request gives 404 manually through telnet
```
# telnet 10.0.1.71 80
Trying 10.0.1.71...
Connected to 10.0.1.71.
Escape character is '^]'.
HEAD /image.jpg HTTP/1.1

HTTP/1.1 404 Not Found
Content-Length: 38

<html>
        <h1>404 Not Found</h1>
</html>Connection closed by foreign host.
```

knnniggett, 2015-04-28:
Seems a "feature" of github is that no one, not even an admin, can delete or remove an issue.  Once it is created, it is always there.  However, you can close it.  Just click the close bottom at the bottom.

jakobmetzger, 2015-04-28:
nvm, still looking into this

closed 2015-04-28 by jakobmetzger