ZoneMinder/zoneminder #828: Missing function drop down values (db issue?)

samaklis, 2015-05-01:
While editing a monitor on the web interface I got a SQL error and now the monitor function dropdown does not get populated. This is what I see on the page source. Any ideas how to fix this?

            <tr><td>Function</td><td><select name="newMonitor[Function]">
SQL-ERR dbFetchOne no result, statement was 'describe Monitors Function'

samaklis, 2015-05-01:
I did a server reboot and the issue went away. Interesting... Still something to look into?

knnniggett, 2015-05-10:
This is likely a configuration issue with your system.  You should check the following:
```
   IMPORTANT: mariadb defaults to strict mode of operation which will cause 
   some zoneminder database writes to fail.  In order to turn this off - 
   which will be for the whole database - you will need to edit 
   /etc/my.cnf and comment out the record 
   
   sql_mode=NO_ENGINE_SUBSTITUTION,STRICT_TRANS_TABLES and restart the
   server
```

closed 2015-06-16 by SteveGilvarry