ZoneMinder/zoneminder #832: Purchase a Raspberry Pi 2 + PiCam for ZoneMinder development

knnniggett, 2015-05-06:
After discussion with @kylejohnson we determined this was the best way to go about purchasing hardware for ZoneMinder development.

This issue has been opened to request the purchase of a Raspberry Pi 2 Kit w/ PiCam using Bountysource funds.

http://www.newegg.com/Product/Product.aspx?Item=9SIA5YB2MP6094&cm_re=raspberry_pi_2-_-9SIA5YB2MP6094-_-Product

The immediate goal of this purchase is to get a ZoneMinder RPM for arm7 architecture on Fedora into zmrepo.  Support for the PiCam will be added at a later date. @connortechnology has already begun development for PiCam support.

kylejohnson, 2015-05-06:
Might need to create a second issue, one for each of you.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

connortechnology, 2015-05-07:
I don't need one.

knnniggett, 2015-05-07:
Someone else want to close this out to "bless" this transaction?

closed 2015-05-07 by connortechnology