ZoneMinder/zoneminder #835: Are you switching to a totally CMake based build?

DanTheColoradan, 2015-05-11:
If so, I have some experience with CMake and I would love to help out if I can during the transition.

knnniggett, 2015-05-11:
Yes, we have already switched. All the rpm's in zmrepo are built using cmake.
The Ubuntu packages are still built using autotools, but I just need to keep prodding @connortechnology to find the time to convert his build process over.  Once that happens we will eventually depreciate autotools.

DanTheColoradan, 2015-05-11:
Awesome. Would it be ok for me to fork, organize it a bit, and  submit a pull? I was thinking of working on that or potentially #108 wiki migration.

DanTheColoradan, 2015-05-12:
If it helps any, I was planning on following a setup like [Open Chemistry](https://github.com/OpenChemistry/openchemistry).

knnniggett, 2015-05-14:
Thank you for your offer to help.
Good documentation is something we always need. 
@SteveGilvarry was the last person I recall who did a significant amount of work with our documentation.  Maybe we can get him to tell us where he left off and what could be done next.  I think we ended up putting all our stuff on Readthedocs, rather than the GitHub Wiki (although both seem to be integrated into github): http://zoneminder.readthedocs.org/en/latest/

On a different topic, if you are willing, there is an old issue building zoneminder that I could use some help with. When building zoneminder with cmake, it will check for and find the ffmpeg devel headers files, but when attempting to build, the compiler does not see them.  The end user's build output looks like this: #304 

The work around, is to just link the development headers up one folder like so:
```bash
ln -s /usr/include/ffmpeg/libavcodec /usr/include/libavcodec
ln -s /usr/include/ffmpeg/libavfilter /usr/include/libavfilter
ln -s /usr/include/ffmpeg/libavutil /usr/include/libavutil
ln -s /usr/include/ffmpeg/libswresample /usr/include/libswresample
ln -s /usr/include/ffmpeg/libavdevice /usr/include/libavdevice
ln -s /usr/include/ffmpeg/libavformat /usr/include/libavformat
ln -s /usr/include/ffmpeg/libpostproc /usr/include/libpostproc
ln -s /usr/include/ffmpeg/libswscale /usr/include/libswscale
```
Since I've got a workaround, I have not taken the time to determine where the problem lies, but I'd really like to resolve this.
Is cmake not setting something correctly?
Is this a packaging problem with the ffmpeg the rpmfusion repo?
Even if the root cause is not with cmake, can we put a workaround in the CmakeLists.txt file?

Let me know if you need anything else.

mastertheknife, 2015-05-15:
Hi Andrew,

Its already implemented. we search for the header files and add the
directory to our include path.
```
find_path(SWSCALE_INCLUDE_DIR "libswscale/swscale.h")  if
(SWSCALE_INCLUDE_DIR)  include_directories("${SWSCALE_INCLUDE_DIR}")
set(CMAKE_REQUIRED_INCLUDES
"${SWSCALE_INCLUDE_DIR}")  endif(SWSCALE_INCLUDE_DIR)
```

Something is probably not working right.

DanTheColoradan, 2015-05-19:
I'm more than willing to look into this. Thanks! I'll fork and work on master branch.

SteveGilvarry, 2015-05-19:
Just to respond to the documentation question above, I still believe #108 is a valid action. I actually started doing it, github wiki is a weird repo, but I have it with a rip of the mediawiki content loaded. Not sure where I got to and the wiki probably needs another rip to update the content. Wiki content is very different to documentation. Continue in #108 if interested in what I did I can PR it.

DanTheColoradan, 2015-06-06:
I've done a bit of work on cmake - nearly all of it was organizing things so they were easier for me (and hopefully others) to understand.

https://github.com/DanTheColoradan/ZoneMinder/tree/cmake/cleanup

It builds using docker and no extra options, so I feel like I can now move forward on making *actual* changes.

Let me know if you have any thoughts.

DanTheColoradan, 2015-06-06:
Could I make my cmake version as sparse as possible?

I imagine supporting very few things to start (e.g. openssl or gnutls - not either or both). Is that too much?

I'm a newb programmer, in relative terms, but I'm trying to take the wikipedia approach and be bold so... I feel like the ZM codebase has a some older use-cases it's supposed to work with, or corner cases, and that might make the codebase 'big' (by big I mean code size to volunteer programmer ratio).

I would love to see the core of ZM support modern architecture and just keep up with that. I feel like that's the directions ZM might be taking, and if that's the case, I suppose I'm really looking for confirmation that I'm doing things that won't just bit rot.

Thanks for any advice. I have used a couple of closed source CCTV programs and have been quite let down. If I can contribute to ZM and make software we are all happy with, I really want.

Thank you for any direction :dancer:

knnniggett, 2015-06-08:
@DanTheColoradan I just got back from a long vacation and am starting the process of catching up on my messages.  Looks like you've created a new branch in your fork of zoneminder to contain your proposed changes, rather than work from your master branch (+1 for that).  However, you need to take it one step further by generating a pull request.  

Look near the top of your github web portal that loosely says something like "You recently pushed changes, would you like to generate a pull request?"  In which case, just push the button to generate a pull request.  This will allow us to easily see what changes you intend to make, and allow us to have a discussion about it.

If you don't see that message, then generating a pull request is just a few extra mouse clicks:
https://help.github.com/articles/creating-a-pull-request/

In the description, make reference to this issue.

bill-mcgonigle, 2015-06-24:
I needed to fix the ffmpeg header case to get the el7 rpm building.  Changes are at ec1c3cfed0178189fed625ec56ec153c98cfc1a4 .
As I understand the cmake documentation, it will still look in the default paths before searching the extra paths listed as hints, so this shouldn't break distros that put ffmpeg headers in /usr/include/ directly.  
Looking at complaints on the issue tracker, it looks like debian* and fedora* put them in /usr/include/ffmpeg/ so I'm not sure which ones don't and haven't tested any here.  If somebody has such a buildbox handy, it would be great to have confirmation.

DanTheColoradan, 2015-06-25:
I can definitely submit a pull request. The barrier for me right now is I have no idea how competent I am to make the changes I'm thinking about. I'll probably keep playing around and learning more before I submit a pull request.

I don't want to hijack the project or screw things up long term. I want contribute in a refined way. My biggest concern right now is writing code that is succinct at the expense of supporting old platforms... And I'm not entirely sure that's where the community wants to head.

Thanks for the responses. I love ZoneMinder and will learn no matter the outcome.

knnniggett, 2015-06-25:
@DanTheColoradan commit ec1c3cf fixes the build issue with ffmpeg header files under /usr/include/ffmpeg.  I built master with this commit and can confirm it works as intended.  So we can mark that one as completed.

Generally speaking, you can expect us to support the two most recent releases of a particular distro.  We try to apply that general rule to the *BSD's as well.  Key word is try.  

Don't worry about breaking things with the project.  Since you don't have write permission to the main repo (you've only got write permission to your own fork), there really isn't anything you can do to break things.  When you create a pull request, it is up to us, the maintainers, to review your work and then merge it.  The responsibility is on us to either bless what you did or suggest changes.

As practice, feel free to generate a pull request.  Anytime you create a PR that you know isn't ready or shouldn't be merged, then just say so in your comments.

closed 2015-07-08 by DanTheColoradan

DanTheColoradan, 2015-08-09:
Thanks Knnniggett. I'm starting grad school next week and that's why I closed this issue. There are a lot of projects I'm interested in and Zoneminder is at the top (the others are less reality based like Minecraft modding). Zoneminder is, however, a huge project O_O and I'm starting to read about appropriate forking (https://jamesdixon.wordpress.com/forking-protocol-why-when-and-how-to-fork-an-open-source-project/). 

I was starting to think 'just fork and make a libvlc only zoneminder'. Now that I've tried it though I know that libvlc is a resource hog. So, I'm still figuring out how I can help out. It's just such a big project and I don't want to break all the things you all have done.

My primary languages are C++ and perl. I would love to simplify the heck out of zone minder, but I really don't have much idea as to how. CMake was what I was originally thinking, so perhaps that? As part of that, I would really love to just cut the heck out of the branches and get down to a gitflow (https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow) branch system. Right now there are (understably) a lot of branches, and - that for me - makes it hard to contribute as I feel trepid about my patches / changes. It goes the same for issues. There are so many issues still open that I feel flummoxed about whether I have anything to contribute.

In any case, I'll keep looking. I would love to just break the project down into very very simple C++ pieces... But given its magnitude, I'm at a lost. If I can somehow fit into you all's *opus magnum*, I want to do it in a way that just simplifies everything. If one of you can give me small goals to do that, I'll get them done.

Thanks for this wonderful body of work, and :beers: to you all. Even if I can't contribute, I'm still using ZM right now and it works damn well.

connortechnology, 2015-08-09:
@DanTheColoradan Generally what you want to do is fork here on github, then you work in your own copy.  This isn't the same as forking a project, it just gives you your own place to play.  

ZoneMinder needs a lot of auditing/cleanup.  Perhaps I can help guide you in where to jump in.  How good is your C++?  Cuz mine is hella rusty.  

I would say jump in wherever it pleases you to do so.  If that is the vlc code, then great.  We have some problems with the vlc camera interface.  Same with the ffmpeg.  So to be more clear, the Camera object/api needs better definition, and more use of inherited code/etc.

knnniggett, 2015-08-09:
@DanTheColoradan Within this context, when we refer to forking, we are talking about simply hitting the Fork button on the top right of your github web portal.  This does what @connortechnology mentioned. It gives you a means to modify the code and then eventually present it back to us in the form of a pull request.

The article you referenced is something different entirely.  In that context, you are forking to develop an entirely new application. 

We are not the original authors of zoneminder, and I don't think anyone, myself especially, can say they have an excellent understanding of the entire project. To some degree I'm sure we all have had the feeling of not knowing how or where to start. 

The way to overcome that is to just stay involved, even if that means casual conversation.  Those of us who remain active in the project all converse on the ZoneMinder IRC channel.  It is where we pass around ideas. Our github forum is not the desired place for general discussion.  I'd encourage you to hang out on IRC.  It will give you a sense of who is working on what and what kind of problems we face.  In time, you'll find a way to plug in.

Reflecting back on my own college experience including graduate school, two of the most important things I learned were not directly related to the classes I was taking.  Instead they were 1) how to break down large problems into smaller, more manageable pieces, and 2) it often takes time, sometimes a lot of time.  Just because you've spent an hour, a day, a week, or possibly longer on something and have not arrived at a solution does not necessarily mean you should stop.  For me, Calculus seemed impossibly difficult at first. After several years it did sink in and things felt easier, even though in reality the school work was more complicated. Note that it took years.  I'd argue that  patience and perseverance are more important than raw smarts.

Today, when I'm faced with a problem I can't immediately solve, I know to just hang in there. So if this is where you feel you are at when looking at the ZoneMinder project, my advice to you is to just hang in there.

DanTheColoradan, 2015-08-09:
Thanks guys.

I find it hard to judge how competent I am at C++, but I would say I have modest abilities now.

I'll go ahead and move my conversation to IRC.