ZoneMinder/zoneminder #839: After initial setup, unable to start again

projectseven, 2015-05-15:
Hello, 
When starting zoneminder we receive the following error in log:

2015-05-15 08:38:15.542690	zmpkg	1938	ERR	Unable to run "/usr/bin/zmdc.pl startup", output is "Starting server"	zmpkg.pl	
2015-05-15 08:38:15.514140	zmdc	1946	FAT	Can't connect: No such file or directory	zmdc.pl	
2015-05-15 08:38:07.611470	zmdc	1949	FAT	Can't bind: Permission denied	zmdc.pl	
2015-05-15 08:38:05.518020	zmdc	1949	INF	Server starting at 15/05/15 08:38:05	zmdc.pl	
2015-05-15 08:38:05.260390	zmpkg	1938	INF	Command: start	zmpkg.pl

Installation was done on Ubuntu Server 14.04 following this guide http://www.zoneminder.com/wiki/index.php/Ubuntu_Server_14.04_64-bit_with_Zoneminder_1.28.1_the_easy_way

What have I missed? Any help would be... helpful! 

Thank you

knnniggett, 2015-05-15:
Can't bind = can't create a socket in the rundir fodler
Please show exactly how you are attempting to start zoneminder.

projectseven, 2015-05-15:
Thank you for the fast reply. By the web GUI, at the top. Is there a better way?

knnniggett, 2015-05-15:
Zoneminder is intended to be initially started via init script:
```
sudo service start zoneminder
```

Zoneminder runs under the web-account user, which is by design an unprivileged account.  Zoneminder relies on several folders to be created on the filesystem before it can run: tmpdir, sockdir, rundir, etc.  However, these folders require an admin account to be created.

TLDR:
Always start zoneminder via the system init script for the first time after bootup.  Following that, it is safe to start/stop/restart zoneminder all you want from the gui.

projectseven, 2015-05-15:
When I sudo service start zoneminder I get start: unrecognized service returned

knnniggett, 2015-05-15:
sorry got it backwards:
```
sudo service zoneminder start
```

projectseven, 2015-05-15:
I hate to be so complicated! That returned: /etc/init.d/zoneminder: 95: /etc/init.d/zoneminder: Syntax error: end of file unexpected (expecting "}")

davidsainty, 2015-05-15:
I struck this error today too, for different (self-inflicted) reasons.  The
error message really could be a lot more helpful - I had to strace and
analyse in order to find the permissions problem.  Logging the socket path
the error occurred on would make diagnosis far more straightforward.

knnniggett, 2015-05-15:
The guide you followed mentions editing the file in question.  
It sounds like you fat-fingered something.

projectseven, 2015-05-15:
I am so embarassed. You are correct, the whole add sleep 15 doesnt have the closing }.

Thank you for your time.

closed 2015-06-16 by SteveGilvarry