ZoneMinder/zoneminder #840: Adapt to changes in Ubuntu 15.04

seeebek, 2015-05-15:
There are changes in Ubuntu 15.04 which make the building and installing Zoneminder not trivial.
The two most significant:
1. Ffmpeg is again in the official repository
2. Initd was replaced by systemd

I suggest creating a new directory under distros named "ubuntu1504" since it is significantly different from 12.04. 
I already got Zoneminder (H264 branch) compiled by changing the dependencies to the new ffmpeg packages provided in 15.04. Now I will be working on replacing initd with systemd. Once I finished and tested my changes I will create a PR for you to review.
If you have any suggestions please let me know.

knnniggett, 2015-05-15:
Don't know how familiar you are with systemd, but many of us that work with Fedora & OpenSuse have been doing this for over a year.  We've been able to get ZoneMinder to work well with systemd.
Welcome to the party. :-)

In anticipation of this day, I worked with Bob to put together this guide here:
http://www.zoneminder.com/wiki/index.php/Ubuntu_Server_14.10_64-bit_server_with_Zoneminder_to_use_systemd

I'm sure you know, 14.10 shipped with systemd, but it was not "turned on".
For you, the relevant parts of that guide are the portions that show how to create the systemd service file and configuring the new tempfiles.d service.

seeebek, 2015-05-16:
Thanks. This is what I was intending to do. Just put it all together. I'm also new to systemd so I will "learn by doing".  I can't promise to do it very soon but I will try my best.

seeebek, 2015-05-24:
I made the changes for Ubuntu 15.04. It compiles and runs for me on a clean installation amd64. It would be nice if @connortechnology  could review it and see if I missed something. Especially the file postinst. There are two places I commented where I wasn't completelly sure.

seeebek, 2015-05-25:
@connortechnology 
I ran into problems with file permissions. caused by following entry in file postinst:
```sh
if [ -z "$2" ]; then
	chown www-data:www-data -R /var/cache/zoneminder
fi
```
$2 gives 1.28.1+1-trusty-SNAPSHOT2015030201 with the latest git master which prevents it from setting the right permissions. What is the design idea behind this?

There is also this:
```sh
if [ "$1" = "configure" ]; then
	if [ -z "$2" ]; then
		chown www-data:www-data /var/log/zm
		chown www-data:www-data /var/lib/zm/
		chown www-data:www-data -R /var/cache/zoneminder
	else
		chown www-data:www-data /var/log/zm
		zmupdate.pl
	fi
fi
```
which prevents zmupdate.pl from being run.
In case of release without the snapshot number the file permissions would be set twice. 

In order to make it work for everybody who wants to build the package from git/source ` if [ -z "$2" ]; then` would need to be changed. I don't see any practical reason to prevent the actions in the way it is right now. Is is not also braking the package on ubuntu1204?

connortechnology, 2015-05-25:
I have no idea.

Manupap did a lot of nice work on these packaging scripts in the debian8 folder.  You might want to look there for inspiration.

seeebek, 2015-05-26:
I redid the setup and I don't have the problem any more. It seems I made something wrong on the way. I think #850 is so far ready to be merged.

Later I will try to make the script based on debian8. Manupap did indeed a beautiful work.

seeebek, 2015-06-05:
So far the basic package build script for ubuntu 15.04 works well. I close this issue.
I will open a new one when I start to work on the debconf solution.

closed 2015-06-05 by seeebek