ZoneMinder/zoneminder #841: Poor video with FOSCAM FI8620

christophey2k, 2015-05-17:
Hello all, have this probleme with my older FOSCAM FI8620 this cam support only h264
and work well with libvlc (but libvlc do not work under zm since long time) with ffmpeg have a blue screen and with remote ave a poorly image like this
[image]

Remote RTSP
Remote method  RTP/Unicast (the only setting that cam support with zm)
Remote host Name admin:password@192.168.253.97
Remote Host port 554
Remote Host path /11
Color 32bits
704x576 at 25fps

This bug seem to be corrected for FOSCAM FI9821W that work like a charm now
but not for this FI8620
i try many setting in camera with no success

SteveGilvarry, 2015-06-17:
Interesting as this looks just like the ffmpeg smearing issue we see, which seems to be something UDP related. You might need to do some more debugging for the devs working on remote rtsp. I am also going to close #741 as they appear to be the same camera and issue.

knnniggett, 2015-06-17:
Does the camera image smear immediately or does it do that over time?
In order to even begin troubleshooting this, we need to see how you programmed this camera into zoneminder (show all settings), and a level 5 debug that shows the RTSP handshake (DESCRIBE, STEUP, PLAY, etc).

christophey2k, 2015-06-22:
Hi, yes that occur immediatly and i turn on debug level 5
 (i run a new install and with ffmpeg the video is less "smear" but not stable) with "remote" is inusable i send you pics:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

knnniggett, 2015-06-22:
The information we need to see, is literally going to contain the words OPTIONS, SETUP, DESCRIBE, PLAY, etc in all caps in your log files.  

- Verify your debug level is set to 5 under options
- Clear out the values in your Max FPS and Alarm Max FPS fields
- Set the frame rate *in the camera* to something low, such as ~5fps
- From the moment you save these changes, zoneminder will attempt to initiate an RTSP handshake with camera.  Your logs will contain the information we are looking for at this moment.  We need to see the logs beginning with the rtsp handshake all the way up until you either see "capturing at x.xx fps" or you get an error message.  Do not open a live during this test.

What would help also greatly is if you also configured zoneminder to write to a log file. This is done under Options, at the same place you set the debug level.  Then just paste the log-file text into a response.

christophey2k, 2015-06-22:
Thank for your reply !
Ok, i setup as describe and clean log file.
I run start ZM for less 1 minute and stop with only FI8620 enabled.
This is the file ( i just replace password cam)

06/22/15 14:43:38.349144 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_logger.cpp/233 [LogOpts: level=DB5/DB5, screen=OFF, database=DB9, logfile=DB5->/var/log/zoneminder/zm_debug.log.09581, syslog=OFF]
06/22/15 14:43:38.349988 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
06/22/15 14:43:38.350464 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/2494 [Got 1 monitors]
06/22/15 14:43:38.350743 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/2529 [Got 1 for v4l_captures_per_frame]
06/22/15 14:43:38.351185 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_camera.cpp/39 [New camera id: 3 width: 704 height: 576 colours: 4 subpixelorder: 8 capture: 1]
06/22/15 14:43:38.351938 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_thread.cpp/241 [Creating thread]
06/22/15 14:43:38.352064 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_rtsp.cpp/197 [RTSP Local SSRC is 2ea17d21]
06/22/15 14:43:38.352483 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_thread.cpp/281 [Starting thread]
06/22/15 14:43:38.353391 zmc_m3[9586].DB2-/usr/src/zoneminder/src/zm_thread.cpp/253 [Invoking thread]
06/22/15 14:43:38.353790 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_thread.cpp/302 [Started thread 9586]
06/22/15 14:43:38.354072 zmc_m3[9586].DB2-/usr/src/zoneminder/src/zm_rtsp.cpp/48 [Sending RTSP message: OPTIONS rtsp://192.168.253.97:554/11 RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 1

]
06/22/15 14:43:38.354682 zmc_m3[9581].ERR-/usr/src/zoneminder/src/zm_logger.cpp/607 [Can't insert log entry: Lost connection to MySQL server during query]
06/22/15 14:43:38.354705 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_image.cpp/167 [Blend: Using SSE2 fast blend function]
06/22/15 14:43:38.354710 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_image.cpp/203 [Delta: Using SSSE3 delta functions]
06/22/15 14:43:38.354712 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_image.cpp/247 [Deinterlace: Using SSSE3 delta functions]
06/22/15 14:43:38.354715 zmc_m3[9581].DB2-/usr/src/zoneminder/src/zm_image.cpp/260 [Image buffer copy: Using SSE2 aligned memcpy]
06/22/15 14:43:38.354717 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_image.cpp/267 [Setting up static colour tables]
06/22/15 14:43:38.354728 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/373 [monitor purpose=1]
06/22/15 14:43:38.354730 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/381 [mem.size=81102560]
06/22/15 14:43:38.354740 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/527 [Unable to map file /dev/shm/zm.mmap.3 (81102560 bytes) to locked memory, trying unlocked]
06/22/15 14:43:38.354743 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/530 [Mapped file /dev/shm/zm.mmap.3 (81102560 bytes) to locked memory, unlocked]
06/22/15 14:43:38.361693 zmc_m3[9586].DB2-/usr/src/zoneminder/src/zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 200 OK
Server: HiIpcam/V100R003 VodServer/1.0.0
Cseq: 1
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, GET_PARAMETER

 (135 bytes)]
06/22/15 14:43:38.361753 zmc_m3[9586].DB2-/usr/src/zoneminder/src/zm_rtsp.cpp/48 [Sending RTSP message: DESCRIBE rtsp://192.168.253.97:554/11 RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 2

]
06/22/15 14:43:38.374171 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/436 [Monitor FOSCAM-FI8620-TERRASSE has no zones, adding one.]
06/22/15 14:43:38.374200 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_zone.cpp/51 [Initialised zone 0/All - 1 - 704x576 - Rgb:0000ff, CM:3, MnAT:15, MxAT:0, MnAP:50, MxAP:75000, FB:3x3, MnFP:50, MxFP:50000, MnBS:10, MxBS:0, MnB:0, MxB:0, OF: 0, AF: 0]
06/22/15 14:43:38.375562 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/446 [Monitor FOSCAM-FI8620-TERRASSE has function 2]
06/22/15 14:43:38.375568 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/447 [Monitor FOSCAM-FI8620-TERRASSE LBF = '%N - %d/%m/%y %H:%M:%S', LBX = 0, LBY = 0]
06/22/15 14:43:38.375570 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_monitor.cpp/448 [Monitor FOSCAM-FI8620-TERRASSE IBC = 50, WUC = 25, pEC = 25, PEC = 25, EAF = 1, FRI = 1000, RBP = 6, ARBP = 6, FM = 0]
06/22/15 14:43:38.375815 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_zone.cpp/957 [Got 1 zones for monitor FOSCAM-FI8620-TERRASSE]
06/22/15 14:43:38.375825 zmc_m3[9581].DB5-/usr/src/zoneminder/src/zm_zone.cpp/989 [Parsing polygon 0,0 703,0 703,575 0,575]
06/22/15 14:43:38.375828 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/823 [Parsing polygon string '0,0 703,0 703,575 0,575']
06/22/15 14:43:38.375834 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/861 [Got coordinate 0,0 from polygon string]
06/22/15 14:43:38.375836 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/861 [Got coordinate 703,0 from polygon string]
06/22/15 14:43:38.375839 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/861 [Got coordinate 703,575 from polygon string]
06/22/15 14:43:38.375841 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/861 [Got coordinate 0,575 from polygon string]
06/22/15 14:43:38.375843 zmc_m3[9581].DB3-/usr/src/zoneminder/src/zm_zone.cpp/881 [Successfully parsed polygon string]
06/22/15 14:43:38.375846 zmc_m3[9581].DB1-/usr/src/zoneminder/src/zm_zone.cpp/51 [Initialised zone 3/All - 1 - 704x576 - Rgb:530000ff, CM:3, MnAT:12, MxAT:0, MnAP:16220, MxAP:0, FB:3x3, MnFP:12164, MxFP:0, MnBS:8110,  [... cut ...]
06/22/15 14:43:40.931435 zmc_m3[9639].DB5-/usr/src/zoneminder/src/zm_rtp_source.cpp/175 [Packet transit RTP time = 43]
06/22/15 14:43:40.931442 zmc_m3[9639].DB5-/usr/src/zoneminder/src/zm_rtp_source.cpp/193 [RTP Jitter: 0]
06/22/15 14:43:40.931584 zmc_m3[9639].DB4-/usr/src/zoneminder/src/zm_rtp_data.cpp/98 [Got 1038 bytes on sd 5]
06/22/15 14:43:40.931594 zmc_m3[9639].DB5-/usr/src/zoneminder/src/zm_rtp_data.cpp/44 [Ver: 2]
06/22/15 14:43:40.931600 zmc_m3[9639].DB5-/usr/src/zoneminder/src/zm_rtp_data.cpp/45 [P: 0]

knnniggett, 2015-06-22:
Yes, that is what we need.  That information looks good. 

However, it looks like you stopped short.  There is only 2 seconds of data there.  We need everything, starting with what you just posted, all the way up to when errors start happening, with no gaps in between.  Note that, a timeout will take at least 10 seconds to occur, so I'd expect to see at least that much.

Since it looks like there is going to be a lot of information, trying posting the log file into a Gist and reference it here.

christophey2k, 2015-06-22:
ok this is the log file
https://gist.github.com/christophey2k/2157405760bf2d57db0b#file-gistfile1-txt

christophey2k, 2015-06-22:
another log file ...
https://gist.github.com/christophey2k/a8d4834c3f5dd1095810#file-fi8620-foscam-log-file

knnniggett, 2015-06-22:
Ok, so the RTSP handshake is working fine, and it looks like, with the exception of a few bad frames, the stream is working for the most part.  The errors are not fatal, but unfortunately the result is a corrupted view.

The error is being called from here:
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_remote_camera_rtsp.cpp#L310

The "error decoding frame" message gets triggered when avcodec_decode_video returns a negative value.  avcodec_decode_video comes from ffmpeg.

So what version of ffmpeg is on your system?

christophey2k, 2015-06-22:
The latest stable on my gentoo 64 is:
 # ffmpeg 
ffmpeg version 2.6.3 Copyright (c) 2000-2015 the FFmpeg developers
  built with gcc 4.8.4 (Gentoo 4.8.4 p1.6, pie-0.6.1)
  configuration: --prefix=/usr --libdir=/usr/lib64 --shlibdir=/usr/lib64 --mandir=/usr/share/man --enable-shared --cc=x86_64-pc-linux-gnu-gcc --cxx=x86_64-pc-linux-gnu-g++ --ar=x86_64-pc-linux-gnu-ar --optflags=' ' --disable-static --enable-avfilter --enable-avresample --disable-stripping --enable-nonfree --enable-version3 --enable-version3 --disable-indev=v4l2 --disable-outdev=v4l2 --enable-version3 --enable-bzlib --enable-runtime-cpudetect --disable-debug --disable-doc --disable-gnutls --enable-gpl --enable-hardcoded-tables --enable-iconv --enable-lzma --enable-network --enable-openssl --enable-postproc --enable-libsmbclient --enable-ffplay --enable-vaapi --disable-vdpau --enable-xlib --enable-libxcb --enable-libxcb-shm --enable-libxcb-xfixes --enable-zlib --enable-libcdio --disable-libiec61883 --enable-libdc1394 --disable-libcaca --enable-openal --enable-opengl --disable-libv4l2 --enable-libpulse --enable-libopencore-amrwb --enable-libopencore-amrnb --disable-libfdk-aac --enable-libopenjpeg --enable-libbluray --disable-libcelt --disable-libgme --disable-libgsm --disable-libmodplug --disable-libopus --disable-libquvi --disable-librtmp --disable-libssh --disable-libschroedinger --enable-libspeex --enable-libvorbis --enable-libvpx --disable-libzvbi --disable-libbs2b --disable-libflite --disable-frei0r --disable-libfribidi --disable-fontconfig --disable-ladspa --disable-libass --enable-libfreetype --disable-libsoxr --enable-pthreads --enable-libvo-aacenc --disable-libvo-amrwbenc --enable-libmp3lame --disable-libaacplus --disable-libfaac --enable-libtheora --enable-libtwolame --enable-libwavpack --disable-libwebp --enable-libx264 --disable-libx265 --enable-libxvid --disable-amd3dnow --disable-amd3dnowext --disable-fma4 --disable-xop --cpu=generic
  libavutil      54. 20.100 / 54. 20.100
  libavcodec     56. 26.100 / 56. 26.100
  libavformat    56. 25.101 / 56. 25.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 11.102 /  5. 11.102
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  1.100 /  1.  1.100
  libpostproc    53.  3.100 / 53.  3.100
Hyper fast Audio and Video encoder

knnniggett, 2015-06-22:
Observation:
The frames that cause errors are all 8 bytes in size, and occur 2 seconds apart from one another.

christophey2k, 2015-06-22:
i dont know if that can help
but when i try this ffmpeg command with FI8620 @25fps (with zm stopped):
# ffmpeg -i rtsp://admin:password@192.168.253.97:554/11 FI8620.avi

have this, but the video generated is correct 

[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
    Last message repeated 10 times
Input #0, rtsp, from 'rtsp://admin:password@192.168.253.97:554/11':
  Metadata:
    title           : \11
  Duration: N/A, start: 0.040000, bitrate: N/A
    Stream #0:0: Video: h264 (Constrained Baseline), yuv420p, 704x576, 25 fps, 25 tbr, 90k tbn, 180k tbc
Output #0, avi, to 'FI8620.avi':
  Metadata:
    INAM            : \11
    ISFT            : Lavf56.25.101
    Stream #0:0: Video: mpeg4 (FMP4 / 0x34504D46), yuv420p, 704x576, q=2-31, 200 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc56.26.100 mpeg4
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> mpeg4 (native))
Press [q] to stop, [?] for help
Past duration 0.999718 too large
    Last message repeated 3 times
Past duration 0.999443 too large
    Last message repeated 3 times
Past duration 0.999153 too large
    Last message repeated 2 times
Past duration 0.998878 too large
    Last message repeated 3 times
Past duration 0.998604 too large
    Last message repeated 3 times
Past duration 0.998329 too large
    Last message repeated 1 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.998329 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.998329 too large
Past duration 0.998055 too large
    Last message repeated 2 times
Past duration 0.997765 too large
    Last message repeated 1 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.997765 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.997765 too large
Past duration 0.997490 too large
    Last message repeated 2 times
Past duration 0.997490 too large     97kB time=00:00:01.40 bitrate= 570.3kbits/s    
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.997215 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.997215 too large
    Last message repeated 1 times
Past duration 0.996941 too large
    Last message repeated 2 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
    Last message repeated 2 times
Past duration 0.996941 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.996666 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.996666 too large
    Last message repeated 2 times
Past duration 0.996376 too large
Past duration 0.996376 too large    119kB time=00:00:01.92 bitrate= 508.6kbits/s    
    Last message repeated 1 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.996101 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.996101 too large
    Last message repeated 2 times
Past duration 0.995827 too large
    Last message repeated 2 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.995552 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.995552 too large
    Last message repeated 2 times
Past duration 0.995277 too large    148kB time=00:00:02.44 bitrate= 498.1kbits/s    
    Last message repeated 1 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.995277 too large
    Last message repeated 1 times
Past duration 0.994987 too large
    Last message repeated 2 times
Past duration 0.994713 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.994713 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.994713 too large
    Last message repeated 1 times
Past duration 0.994438 too large
    Last message repeated 1 times
Past duration 0.994438 too large    168kB time=00:00:02.96 bitrate= 465.0kbits/s    
    Last message repeated 1 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.994164 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.994164 too large
    Last message repeated 1 times
Past duration 0.993889 too large
    Last message repeated 2 times
Past duration 0.993599 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.993599 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.993599 too large
    Last message repeated 1 times
Past duration 0.993324 too large
Past duration 0.993324 too large    187kB time=00:00:03.48 bitrate= 439.3kbits/s    
    Last message repeated 2 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.993324 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.993050 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
    Last message repeated 2 times
Past duration 0.993050 too large
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.993050 too large
Past duration 0.992775 too large
    Last message repeated 3 times
[rtsp @ 0x8c8520] Too short data for FU-A H264 RTP packet
Past duration 0.992485 too large
    Last message repeated 2 times
frame=  101 fps= 31 q=8.2 Lsize=     208kB time=00:00:04.04 bitrate= 422.4kbits/s    
video:200kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 3.980776%

curlyboi, 2017-01-04:
The biggest problem is that the error cannot be hidden (If I completely disable logging to DB, I still get this). In my camera, the picture is fine, but I get this error everytime I receive a keyframe (I-Frame). But the picture looks OK. When I switch to FFMPEG instead of Remote, the error is gone.

knnniggett, 2017-01-04:
The remote method is no longer being maintained due to lack of resources. It simply does not make sense for us to spend time on this, when much better solutions exist already (ffmpeg & libvlc).

Our documentation has been updated to reflect the status of this remote rtsp method:
http://zoneminder.readthedocs.io/en/latest/userguide/definemonitor.html#remote

Pull requests are still welcome if anyone else wishes to spend their time on this.

closed 2017-01-04 by knnniggett