ZoneMinder/zoneminder #843: Initiate a discussion on event notification

pliablepixels, 2015-05-20:
I've been thinking of implementing a simple framework that makes it easy for developers to implement event notification mechanisms. The role of this framework is simply to report new events. What one does with that report depends on the nature of notification. For apple phones this will map to APNS, for google phones this will be GCM and for something else, a completely different mechanism.

There are two possible ways to do it:

a) Modify ZM source (zm_events.cpp?) to create a new events trigger
b) Implement a DB trigger
c) Implement a new thin perl daemon and connect it to zmdc.pl that directly polls the DB at regulated intervals (similar to many other daemons in ZM code today in scripts/)

a) is messy and b) is platform dependent

c) seems like a neat way to go about things and does not involve any core changes.

So how about we register a new daemon, say zmeventnotification.pl to zmdc which basically reports the Id and Name of the last X events since it last checked?

I've implemented an initial version here for initiating discussions
https://github.com/pliablepixels/zmhacks/blob/master/zmeventnotification.pl

There is a function called processNewEvents (https://github.com/pliablepixels/zmhacks/blob/master/zmeventnotification.pl#L41) which gets passed the last X new events (X is configurable at install time) and it can do whatever it needs with it.

Thoughts?

pliablepixels, 2015-09-02:
Updated event framework daemon (initial version)

a) Uses shared memory polling to get event notifications  -no DB dips
b) Much faster than ZM filter (obviously) due to above reason
c) single threaded, single process. No need for more threads or processes for a light weight loop
d) I don't think this warrants zm_monitor.cpp to be modified to send a signal when a new event is generated - perl and signals seems to have issues based on distros due to it not being re-entrant -- I personally don't think its worth the complexity 
e) Given that there are multiple delivery mechanisms for an event - GCM, APNS, whatever else, and that it requires certificate setup etc - this is something individuals can do themselves. To help this, zmeventnotification.pl opens up a web socket. A client can connect to it and be notified when there is an event (advantage of web socket ==> no polling)

Please review https://github.com/pliablepixels/zmhacks/blob/master/zmeventnotification.pl

stjohnjohnson, 2016-02-09:
So what are our options here?  Do we have any kind of event notifications currently?

pliablepixels, 2016-02-09:
https://github.com/pliablepixels/zmeventserver

kylejohnson, 2016-02-09:
I'm partial to option a as it allows for real-time notifications (not having to wait for the next daemon loop).  I'd love to have a generic webhook that can fire off as soon as an event happens.

closed 2017-03-18 by SteveGilvarry