ZoneMinder/zoneminder #846: Global symbol requires explicit package name...

hans109h, 2015-05-21:
After build of master today I am unable to run zmupdate

$ zmupdate.pl 
Global symbol "%Config" requires explicit package name at /usr/local/bin/zmupdate.pl line 81.
Global symbol "%Config" requires explicit package name at /usr/local/bin/zmupdate.pl line 81.
Global symbol "%Config" requires explicit package name at /usr/local/bin/zmupdate.pl line 81.
Global symbol "%Config" requires explicit package name at /usr/local/bin/zmupdate.pl line 81.
BEGIN not safe after errors--compilation aborted at /usr/local/bin/zmupdate.pl line 81.

connortechnology, 2015-05-21:
I think you are mixing ilbraries from an old zm with programs from a new zm.

hans109h, 2015-05-21:
So true.  I wonder how long that has been the case....yikes.  Thanks for pointing that out.

closed 2015-05-21 by hans109h