ZoneMinder/zoneminder #848: X-10 configuration not working ZM V1.28.1

peteeee, 2015-05-21:
1 - Steps to install:
perl -MCPAN -e shell
install CPAN
exit
perl -MCPAN -e shell
install YAML PHP::Serialization Module::Load X10::ActiveHome
install Sys::Mmap
exit
2 - connect CM11A via USB to Serial device
3 - adduser www-data dialout 
4 - manually configure X10 monitor check box but it goes away after granular X10 device number configuration save.
mysql -p -u root
USE zm;
UPDATE Monitors SET triggers = 'X10' WHERE id = 1;
UPDATE Monitors SET triggers = 'X10' WHERE id = 2;
UPDATE Monitors SET triggers = 'X10' WHERE id = 3;
UPDATE Monitors SET triggers = 'X10' WHERE id = 4;
exit 

5 - testing via command line: 
zmx10.pl -u 1 -c on 

Bareword "ZM_X10_DEVICE" not allowed while "strict subs" in use at /usr/bin/zmx10.pl line 194.
Bareword "ZM_X10_HOUSE_CODE" not allowed while "strict subs" in use at /usr/bin/zmx10.pl line 194.
Execution of /usr/bin/zmx10.pl aborted due to compilation errors.

knnniggett, 2015-05-21:
Change Line 194 from:
```Perl
$x10 = new X10::ActiveHome( port=>ZM_X10_DEVICE, house_code=>ZM_X10_HOUSE_CODE, debug=>0 );
```

To:
```Perl
$x10 = new X10::ActiveHome( port=>$Config{ZM_X10_DEVICE}, house_code=>$Config{ZM_X10_HOUSE_CODE}, debug=>0 );
```

And then report back.

peteeee, 2015-05-22:
Thank-you for the line change suggestion.
Still an issue:
:~# zmx10.pl -u 1 -c on
Bareword "ZM_X10_HOUSE_CODE" not allowed while "strict subs" in use at /usr/bin/zmx10.pl line 753.
Execution of /usr/bin/zmx10.pl aborted due to compilation errors.

Line 753 reads

if ( $event->house_code() eq ZM_X10_HOUSE_CODE )

changed line 753 to:

if ( $event->house_code() eq $Config{ZM_X10_HOUSE_CODE} )

and got this:

~# zmx10.pl -u 1 -c on
Unable to connect, starting server
Undefined subroutine &X10Server::ZM_X10_DB_RELOAD_INTERVAL called at /usr/bin/zmx10.pl line 207.
        (in cleanup) Can't call method "close" on an undefined value at /usr/local/share/perl/5.18.2/X10/ActiveHome.pm line 434 during global destruction.

Line 207 reads:

 $x10 = new X10::ActiveHome( port=>$Config{ZM_X10_DEVICE}, house_code=>$Config{ZM_X10_HOUSE_CODE}, debug=>0 );

Left it there for now.

closed 2015-06-08 by connortechnology

peteeee, 2015-06-08:
Thank-you!

So I just use the new file eh?

PR #856 (2015-06-08): Correct bareword config entries with newer {} style
Fix #848