ZoneMinder/zoneminder #849: USE_DEEP_STORAGE path makes time zone assumptions

davidsainty, 2015-05-23:
The deep storage path assumes that the database server and ZoneMinder server have the same time zone.  This will possibly not be true if you happen to have changed the time zone on the machine, or happen to run the database server on another machine.

It's also dodgy to use the local time zone at all - if the time zone is changed then the Start Times may go back in time, and any event on the same day and time in the new time zone will overwrite the older event.

Finally, every single year, daylight savings will cause the local time to go backwards, and any event on the time within that daylight savings hour will overwrite the old event.

There are quite a lot of ways to do this more robustly.

Using UTC in both the database data and the file system would be one way, as a monotonically increasing datetime.

You could also use the event ID (like non-deep-style used to) but create sub-directories after every three digits (or two digits) to limit each directory to 1000 (or 100) entries, and not have to worry about time zones or time conversions at all...  E.g. event 12345 would be padded with zeros to satisfy int(10), directory could be 000/000/012/345 or 00/00/01/23/45.

SteveGilvarry, 2015-06-16:
I do like UTC, it would be easier option to implement as opposed to a new file system.

SteveGilvarry, 2017-03-18:
Propose we move to UTC for all server activities and user sets their timezone. Also set camera timezone to control timestamping.

closed 2018-06-22 by stale[bot]