ZoneMinder/zoneminder #852: On ubuntu 14.04 zoneminder 1.28.1 alarm never terminate!!

far5893, 2015-05-26:
Hi,
i use last zoneminder 1.28.1 , and configure motion detect as done for years with precedents versions , but once alarm trigger it stay ON for hours without stopping , sensitivity is correct because allarm start only when needed , but stop only hours from start(and have only long events ).

far5893, 2015-05-26:
OK i answer my self , it's called image blending , and is a cam based option,
but not so clear explained.

knnniggett, 2015-05-26:
@far5893 The zoneminder github forum is not a discussion forum. This is a bug reporting and feature request forum.  Please take this and any future "how do I use zoneminder" type of question to the zoneminder forum at zoneminder.com. Thank you.

closed 2015-05-26 by knnniggett

far5893, 2015-05-26:
Sorry but i think was a bug before discover that it's a feature (Cit.)

knnniggett, 2015-05-26:
Your right.  One thing we can always use is better documentation.  We've got a very large userbase, but we there are just a handful of people willing to give back to this project in the form of writing code, testing, and documentation.  You can help by updating the documentation found in the WiKi.