ZoneMinder/zoneminder #855: zm_signal.cpp ignores signal safety

davidsainty, 2015-05-27:
Both zm_hup_handler() and zm_term_handler() call Info(), which calls zm_logger.cpp:logPrint(), which is a long long long way off being signal safe.

The solution has to be to not call the logging functions from the signal handlers.

The reason why I found this is that I have a one-off problem with an event, with Start Time = May 26 18:34:21, but the frames are deep storage'd in 15/05/26/18/34/25.  Because the directory doesn't match, the event is unviewable.

This happened because zm_logger.cpp:logPrint() calls localtime() and so does zm_event.cpp:Event::Event(), and the TERM signal just happened to be called at the right time to trash the contents of the localtime() results it appears.

* 1432640065.372001 | zma_m1 | 22291 |0|INF| Reception: 073 - Gone into alarm state               | zm_monitor.cpp | 1444 | 2015-05-26 18:34:25
* 1432640065.409520 | zmdc   |  2524 |0|INF| 'zma -m 1' stopping at 15/05/26 18:34:25             | zmdc.pl        | NULL | 2015-05-26 18:34:25
* 1432640065.759327 | zma_m1 | 22291 |0|INF| Got signal 15 (Terminated), exiting                  | zm_signal.cpp  |   40 | 2015-05-26 18:34:25
* 1432640065.813068 | zma_m1 | 22291 |0|INF| Reception: 073 - Opening new event 2305, alarm start | zm_monitor.cpp | 1464 | 2015-05-26 18:34:25

On a side note, the call to `Warning( "Event %d has zero time, setting to current", id );` in zm_event.cpp would have the same breaking effect.  This simpler case could be solved by extracting the contents of the localtime() results immediately, rather than assuming they will be safe - that really should happen here.

connortechnology, 2015-05-27:
I've noticed similar interactions with signals in zmaudit.  Not even multi-threaded and we have concurrency issues.

I am starting to have serious concerns with the deep storage method.  

I am starting to think that instead of storing by date.. we should store by event Id and just build a tree based on the digits of the event so event 123456 would be /events/zm/1/2/3/4/5/6 or something.  

So to work around... how about turning off signals while calculating all these time variables and creating the event?

davidsainty, 2015-05-27:
I think the logging is impractical to leave in the signal handlers at all.  It isn't just localtime(), all the stdio calls, syslog, none of that is safe from a signal handler.

If the signal handlers didn't do logging directly, the deep storage localtime() calls are otherwise trivial to make safe.

[ But, that said, there are plenty of other good reasons to ditch local time-based deep storage and do something like the event ID based deep storage :) ]

closed 2018-06-22 by stale[bot]