ZoneMinder/zoneminder #860: MJPEG rtsp stream not working with Edimax IR-113E

JoelLinn, 2015-05-30:
I can only use the MPEG/264 stream with this camera using the remote option.
Zoneminder has trouble isolating the frame from the stream
[image]
[image]

connortechnology, 2015-06-08:
The demo site doesn't work.  We will need access to a camera to test with.

Are you able to provide remote access to your camera?

SteveGilvarry, 2015-06-16:
@JoelLinn any update on providing a camera someone can test?

knnniggett, 2015-06-22:
Have you tried different a remote method? I've had the best results using RTSP/unicast.
Note that using 32 bit colorspace will give a performance improvement.  

If you continue to have issues, then we will need a complete level 5 debug session beginning with the RTSP handshake leading all the way up to any errors that show up.  #841 shows  an example of what we need to see.

JoelLinn, 2015-06-22:
I have a log sample for you. Password is zoneminder 
https://wolke.leuchtkanone.de/index.php/s/smz7LHEIJ5RWekA
Action starts after 2015-06-22 20:56:00.038760
If that doesn't help I'm happy to provide an OpenVPN tunnel to the camera in question.

knnniggett, 2015-06-22:
No, that does not contain what we need.  Please refer to #841 and provide the same debug level 5 information for your camera.  Raw data is preferred, no xml.

JoelLinn, 2015-06-23:
Interestingly no monitor works at all if i increase the debug level. The capture just crashes

2015-06-23 08:16:58.058510 zmdc[15469].INF-zmdc.pl [Starting pending process, zmc -m 5]
2015-06-23 08:16:58.066700 zmdc[15469].INF-zmdc.pl ['zmc -m 5' starting at 15/06/23 08:16:58, pid = 15627]
2015-06-23 08:16:58.067130 zmdc[15627].INF-zmdc.pl ['zmc -m 5' started at 15/06/23 08:16:58]
2015-06-23 08:16:58.097908 zmc_m5[15627].DB1-zm_logger.cpp/230 [LogOpts: level=DB5/DB5, screen=OFF, database=DB9, logfile=DB5->/var/log/zm/zm_debug.log.15627, syslog=DB9]
2015-06-23 08:16:58.099443 zmc_m5[15627].DB1-zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
2015-06-23 08:16:58.100190 zmc_m5[15627].DB1-zm_monitor.cpp/2486 [Got 1 monitors]
2015-06-23 08:16:58.100817 zmc_m5[15627].DB1-zm_monitor.cpp/2521 [Got 1 for v4l_captures_per_frame]
2015-06-23 08:16:58.101837 zmc_m5[15627].DB2-zm_camera.cpp/39 [New camera id: 5 width: 640 height: 480 colours: 3 subpixelorder: 6 capture: 1]
2015-06-23 08:16:58.104300 zmc_m5[15627].DB1-zm_thread.cpp/241 [Creating thread]
2015-06-23 08:16:58.104994 zmc_m5[15627].DB2-zm_rtsp.cpp/197 [RTSP Local SSRC is 3ec1b975]
2015-06-23 08:16:58.106024 zmc_m5[15627].DB1-zm_thread.cpp/281 [Starting thread]
2015-06-23 08:16:58.106589 zmc_m5[15629].DB2-zm_thread.cpp/253 [Invoking thread]
2015-06-23 08:16:58.107135 zmc_m5[15627].DB1-zm_thread.cpp/302 [Started thread 15629]
2015-06-23 08:16:58.107689 zmc_m5[15627].DB2-zm_image.cpp/167 [Blend: Using SSE2 fast blend function]
2015-06-23 08:16:58.108204 zmc_m5[15629].DB2-zm_rtsp.cpp/48 [Sending RTSP message: OPTIONS rtsp://ir-113e:554/stream2 RTSP/1.0
User-Agent: ZoneMinder/1.28.0
CSeq: 1

]
2015-06-23 08:16:58.108210 zmc_m5[15627].DB2-zm_image.cpp/203 [Delta: Using SSSE3 delta functions]
2015-06-23 08:16:58.112350 zmdc[15469].ERR-zmdc.pl ['zmc -m 5' exited abnormally, exit status 255]
2015-06-23 08:17:06.144330 zmdc[15469].WAR-zmdc.pl [Can't find process with command of 'zma -m 5']
2015-06-23 08:17:06.284530 zmdc[15469].INF-zmdc.pl [Command 'zmc -m 5' removed from pending list at 15/06/23 08:17:06]
2015-06-23 08:17:06.424220 zmdc[15469].WAR-zmdc.pl [Can't find process with command of 'zma -m 5']
2015-06-23 08:17:06.978368 web_php[11534].DBG-includes/logger.php/168 [LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=DBG->/var/log/zm/zm_debug.log.11534, weblog=INF, syslog=INF]
2015-06-23 08:17:07.168418 web_php[13359].DBG-includes/logger.php/168 [LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=DBG->/var/log/zm/zm_debug.log.13359, weblog=INF, syslog=INF]
2015-06-23 08:17:07.170045 web_js[13359].DBG-includes/logger.php/168 [LogOpts: level=DBG/DBG, screen=OFF, database=DBG, logfile=DBG->/var/log/zm/zm_debug.log.13359, weblog=OFF, syslog=DBG]
2015-06-23 08:17:07.171417 web_js[13359].ERR-skins/classic/js/skin.js/164 [Uncaught TypeError: Cannot read property 'focus' of undefined]

knnniggett, 2015-06-23:
From the debug, it looks like you are running zoneminder 1.28.0.  Before we go any further, please upgrade to 1.28.1.  1.28.1 has improvements to the RTSP logic.

JoelLinn, 2015-06-25:
I updated to 1.28.1 from this repo and this is what I get with debug level 5

2015-06-25 19:28:52.744350 zmdc[7516].WAR-zmdc.pl [Can't find process with command of 'zma -m 5']
2015-06-25 19:28:52.884240 zmdc[7516].WAR-zmdc.pl [Can't find process with command of 'zmc -m 5']
2015-06-25 19:28:53.034850 zmdc[7516].INF-zmdc.pl ['zmc -m 5' starting at 15/06/25 19:28:53, pid = 8070]
2015-06-25 19:28:53.035040 zmdc[8070].INF-zmdc.pl ['zmc -m 5' started at 15/06/25 19:28:53]
2015-06-25 19:28:53.070944 zmc_m5[8070].DB1-zm_logger.cpp/233 [LogOpts: level=DB5/DB5, screen=OFF, database=DB9, logfile=DB5->/var/log/zm/zm_debug.log.08070, syslog=DB9]
2015-06-25 19:28:53.072540 zmc_m5[8070].DB1-zm_utils.cpp/262 [Detected a x86\x86-64 processor with SSSE3]
2015-06-25 19:28:53.073972 zmc_m5[8070].DB1-zm_monitor.cpp/2505 [Got 1 monitors]
2015-06-25 19:28:53.075232 zmc_m5[8070].DB1-zm_monitor.cpp/2540 [Got 1 for v4l_captures_per_frame]
2015-06-25 19:28:53.075878 zmc_m5[8070].DB2-zm_camera.cpp/39 [New camera id: 5 width: 640 height: 480 colours: 3 subpixelorder: 6 capture: 1]
2015-06-25 19:28:53.079461 zmc_m5[8070].DB1-zm_thread.cpp/241 [Creating thread]
2015-06-25 19:28:53.080627 zmc_m5[8070].DB2-zm_rtsp.cpp/197 [RTSP Local SSRC is 62887b2]
2015-06-25 19:28:53.081129 zmc_m5[8070].DB1-zm_thread.cpp/281 [Starting thread]
2015-06-25 19:28:53.082643 zmc_m5[8075].DB2-zm_thread.cpp/253 [Invoking thread]
2015-06-25 19:28:53.083263 zmc_m5[8070].DB1-zm_thread.cpp/302 [Started thread 8075]
2015-06-25 19:28:53.084010 zmc_m5[8070].DB2-zm_image.cpp/167 [Blend: Using SSE2 fast blend function]
2015-06-25 19:28:53.084672 zmc_m5[8070].DB2-zm_image.cpp/203 [Delta: Using SSSE3 delta functions]
2015-06-25 19:28:53.085208 zmc_m5[8075].DB2-zm_rtsp.cpp/48 [Sending RTSP message: OPTIONS rtsp://ir-113e:554/stream2 RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 1

]

which seems to differ from what happens without debug:

2015-06-25 19:36:48.110690 zmpkg[8832].INF-zmpkg.pl [Command: start]
2015-06-25 19:36:48.339472 web_js[8520].ERR-http://cctv.privat.linn.local/skins/classic/js/skin.js/117 [Uncaught TypeError: Cannot read property 'focus' of undefined]
2015-06-25 19:36:48.466420 zmdc[8861].INF-zmdc.pl [Server starting at 15/06/25 19:36:48]
2015-06-25 19:36:51.628270 zmdc[8861].INF-zmdc.pl ['zmfilter.pl' starting at 15/06/25 19:36:51, pid = 8890]
2015-06-25 19:36:51.628830 zmdc[8890].INF-zmdc.pl ['zmfilter.pl' started at 15/06/25 19:36:51]
2015-06-25 19:36:51.788840 zmdc[8861].INF-zmdc.pl ['zmaudit.pl -c' starting at 15/06/25 19:36:51, pid = 8901]
2015-06-25 19:36:51.789120 zmdc[8901].INF-zmdc.pl ['zmaudit.pl -c' started at 15/06/25 19:36:51]
2015-06-25 19:36:51.876840 zmfilter[8890].INF-zmfilter.pl [Scanning for events]
2015-06-25 19:36:51.951660 zmdc[8861].INF-zmdc.pl ['zmwatch.pl' starting at 15/06/25 19:36:51, pid = 8912]
2015-06-25 19:36:51.952410 zmdc[8912].INF-zmdc.pl ['zmwatch.pl' started at 15/06/25 19:36:51]
2015-06-25 19:36:52.083290 zmwatch[8912].INF-zmwatch.pl [Watchdog starting]
2015-06-25 19:36:52.085530 zmwatch[8912].INF-zmwatch.pl [Watchdog pausing for 30 seconds]
2015-06-25 19:36:52.113490 zmdc[8861].INF-zmdc.pl ['zmupdate.pl -c' starting at 15/06/25 19:36:52, pid = 8923]
2015-06-25 19:36:52.114110 zmdc[8923].INF-zmdc.pl ['zmupdate.pl -c' started at 15/06/25 19:36:52]
2015-06-25 19:36:52.616040 zmaudit[8901].INF-zmaudit.pl [Database event '5/34788' does not exist in filesystem but too young to delete.]
2015-06-25 19:36:53.992150 zmaudit[8901].INF-zmaudit.pl [Deleted 118 log table entries by time]
2015-06-25 19:36:59.504255 web_js[8600].ERR-http://cctv.privat.linn.local/skins/classic/js/skin.js/117 [Uncaught TypeError: Cannot read property 'focus' of undefined]
2015-06-25 19:37:03.674990 zmdc[8861].WAR-zmdc.pl [Can't find process with command of 'zma -m 5']
2015-06-25 19:37:03.813390 zmdc[8861].WAR-zmdc.pl [Can't find process with command of 'zmc -m 5']
2015-06-25 19:37:03.955930 zmdc[8861].INF-zmdc.pl ['zmc -m 5' starting at 15/06/25 19:37:03, pid = 8987]
2015-06-25 19:37:03.956080 zmdc[8987].INF-zmdc.pl ['zmc -m 5' started at 15/06/25 19:37:03]
2015-06-25 19:37:03.996250 zmdc[8861].ERR-zmdc.pl ['zmc -m 5' exited abnormally, exit status 255]
2015-06-25 19:37:03.998100 zmdc[8861].INF-zmdc.pl [Starting pending process, zmc -m 5]
2015-06-25 19:37:04.004690 zmdc[8861].INF-zmdc.pl ['zmc -m 5' starting at 15/06/25 19:37:04, pid = 8993]
2015-06-25 19:37:04.006070 zmdc[8993].INF-zmdc.pl ['zmc -m 5' started at 15/06/25 19:37:04]
2015-06-25 19:37:04.099950 zmdc[8861].WAR-zmdc.pl [Can't find process with command of 'zma -m 5']
2015-06-25 19:37:04.136475 zmc_m5[8993].INF-zmc.cpp/212 [Starting Capture version 1.28.1]
2015-06-25 19:37:04.817265 web_js[5268].ERR-http://cctv.privat.linn.local/skins/classic/js/skin.js/117 [Uncaught TypeError: Cannot read property 'focus' of undefined]
2015-06-25 19:37:05.042847 zmc_m5[8996].INF-zm_rtsp.cpp/395 [Received new Content-Base in DESCRIBE response header. Updated device Url to: 'rtsp://10.50.1.4/stream2/']
2015-06-25 19:37:05.053997 zmc_m5[8996].WAR-zm_rtp_source.cpp/69 [The device is using a codec that may not be supported. Do not be surprised if things don't work.]
2015-06-25 19:37:05.139520 zmc_m5[8993].ERR-zm_remote_camera_rtsp.cpp/310 [Error while decoding frame 0]
2015-06-25 19:37:05.140724 zmc_m5[8993].ERR-zm_remote_camera_rtsp.cpp/311 [256: 00 00 00 00 41 28 50 3c 00 54 ff ff 43 6c 93 96 c0 08 41 eb dc fe 14 47 03 40 d8 1b 48 65 f9 b7 73 4e 59 01 77 cb b6 3b 64 93 8a 8e e6 5f 95 4e 49 ed c5 73 b [... cut ...]  4d 34 cc 36 9e 40 18 e0 e0 0a 24 db 02 c6 15 36 fc db 54 91 93 81 9c 67 9e 6a 95 ea 85 61 86 76 e4 8f 99 89 c0 c2 9f eb 57 ed db 2f e6 3e 7a 67 6f e5 cf 14 c9 82 bc 6d ca 16 c6 71 e4 b6 73 81 ef ec 3b 56 94 9a b1 2c ca 51 b8 e0 63 fc 29 cc 47 6e 15 46 06 7f 9d 4d 72 59 58 6f 39 66 1c 90 3b 67 fc fe 94 d8 6c da ea 33 23 12 b1 74 18 ea 4e 71 5b 27 71 15 0b 79 92 01 fc 3c ff 00 2a 90]
2015-06-25 19:37:05.467299 zmc_m5[8993].ERR-zm_remote_camera_rtsp.cpp/310 [Error while decoding frame 0]
and so on

closed 2017-03-18 by SteveGilvarry