ZoneMinder/zoneminder #861: repeated zma crashing

bhuddah, 2015-05-30:
i have those errors happen repeatedly.
please have a look:

```
May 30 20:07:24 ubuntukas zma_m5[28698]: INF [Cafe: 104 - Gone into alarm state]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Got signal 11 (Segmentation fault), crashing]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Signal address is 0x63f, from 0x41d337]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 0: /usr/bin/zma() [0x48001f]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x10340) [0x7f452d5cc340]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 2: /usr/bin/zma() [0x41d337]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 3: /usr/bin/zma() [0x449ac3]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 4: /usr/bin/zma() [0x4086a5]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f452c063ec5]]
May 30 20:07:24 ubuntukas zma_m5[28698]: ERR [Backtrace 6: /usr/bin/zma() [0x408199]]
May 30 20:07:24 ubuntukas zma_m5[28698]: INF [Backtrace complete, please execute the following command for more information]
May 30 20:07:24 ubuntukas zma_m5[28698]: INF [addr2line -e /usr/bin/zma 0x48001f 0x7f452d5cc340 0x41d337 0x449ac3 0x4086a5 0x7f452c063ec5 0x408199]
```

connortechnology, 2015-06-03:
Okay, you say when switching off DB logging.  Problem is, saving the Options page doesn't tell the zma process anything.  I don't see how it could cause a crash.

Also, it doesn't crash for me.  So we need to really narrow this down to exactly what action causes the crash.

What level are you logging to the db at?

bhuddah, 2015-06-03:
i always restarted zm after changing the setting.
i have LOG_LEVEL_DATABASE on fatal now and everything works again.
LOG_LEVEL_FILE on info and the two other loglevels on off.

closed 2017-03-18 by SteveGilvarry