ZoneMinder/zoneminder #864: Skin based body { font-size } should be configuable in options

lingfish, 2015-06-03:
I love the flat CSS style applied to classic.  This is of course personal preference, but I find `font-size: 16px;` to be too large and "toy" like ;)

So every time I upgrade, I have to re-hack that (and changes invariably to `popupSizes`).

It seems that most/all other font sizing is rightly using offsets based on `body`, so it shouldn't be that complex?

connortechnology, 2015-06-03:
So... no we aren't going to add config entries for every size in the UI.  

I do agree that various fonts and paddings are too big and various windows are not correctly sized.  We need to fix those.

lingfish, 2015-06-04:
Hi, what do you mean "config entries for every size in the UI"?

I'm suggesting a integer-only freeform entry that would essentially map to the body CSS field.  Most/all other sizes are relative in percentages to that I think?

Worst case, a small/medium/large "overall" font size?

knnniggett, 2015-08-18:
We don't currently have the resources to address feature requests, which are based solely on a particular user's visual preference.  If you would like to submit a pull request, which creates a new skin or makes the flat skin more flexible, then you are welcome to do so.

closed 2015-08-18 by knnniggett