ZoneMinder/zoneminder #865: Odd window size differences when viewing frame alarm image between Chrome & FF

lingfish, 2015-06-03:
Chrome sizes perfectly, Firefox doesn't.  Everything gets nudged down, and a scrollbar is required.

Best way to show this is with an image:

[image]

connortechnology, 2015-06-03:
I just pushed a fix to master for this.  

The popup it was loading was using the dimensions for an image, but this is the frame page, which has another 20px in height for the prev/next buttons.

lingfish, 2015-06-04:
Nice one, thanks.  A bit painful for me to test (Debian, using testing, silly libturbo hacks etc).

connortechnology, 2015-06-04:
I do have a ppa for debian... needs updating.  I'll see about that today.

SteveGilvarry, 2015-06-16:
@lingfish If you want to manually apply the fix it is in referenced commit. Closing the issue, please feel free to re-open if fix described doesn't work.
https://github.com/ZoneMinder/ZoneMinder/commit/f99c5f36278d2b98bddd42959f1f0663accea2ff

closed 2015-06-16 by SteveGilvarry