ZoneMinder/zoneminder #868: cant add or edit monitor

danilomendes, 2015-06-03:
whenever I try to add a new monitor, the form closes but the new camera doesnt shows up. I've noticed that if i try to open the "new monitor" form again, it comes with the same title of the previous "Monitor - Monitor-97".

Watching the logs, they do the following after I press the "save" button.

```
06/03/2015 16:36:34.531787 zmdc[3116].WAR [Can't find process with command of 'zmtrack.pl -m']
06/03/2015 16:36:34.820641 zmdc[3116].WAR [Can't find process with command of 'zma -m']
06/03/2015 16:36:35.128799 zmdc[3116].WAR [Can't find process with command of 'zmc -m']
06/03/2015 16:36:35.444576 zmdc[3116].WAR [Can't find process with command of 'zmtrack.pl -m']
06/03/2015 16:36:35.763903 zmdc[3116].WAR [Can't find process with command of 'zma -m']
```

I have also noticed that if i try to rename one monitor nothing also happen, and the log says:

```
06/03/2015 16:49:12.574213 zmdc[3116].WAR [Can't find process with command of 'zmtrack.pl -m 2']
```

I am using the v1.28.1 on centOS 6.

Thank you!

danilomendes, 2015-06-03:
I just managed to add new monitors adding them directly into the database table Monitors and restarting the service...

Somebody may help me to get the web interface working properly?

Thank you in advance!

newburns, 2015-06-03:
What MySQL version are you running?
Do you have many errors when trying to record those monitors?

danilomendes, 2015-06-03:
mysql version: 5.6.23

I've noticed that since I added the new monitors, they didnt record anything as modect, but there's not much motion around here at this time.

Changing the monitor's function to "record", it starts to record normally.

newburns, 2015-06-03:
#649 
#672 
You need to turn off the strict_trans_table I think.
Does that resolve your issue?

hrwebasst, 2015-06-03:
Run this as root on your mysql server
SET @@global.sql_mode= '';

danilomendes, 2015-06-04:
thank you! that did the trick! sorry i missed that previous posts!

closed 2015-06-04 by danilomendes

corn13read, 2015-06-24:
adding this to my.cnf may also fix the problem, haven't had a chance to try it yet though:

sql_mode=NO_ENGINE_SUBSTITUTION,STRICT_TRANS_TABLES

danilomendes, 2015-06-24:
@corn13read, that was exactly what I needed to do, so I wont loose this setting if I eventually need to reboot the server.

In my case, i needed to put that on /usr/my.cnf that overrides the /etc/my.cnf file. But I think that this is a centOS particularity.

corn13read, 2015-06-24:
@danilomendes yeah, you could feel free to check out my docker if you'd like, I'm going to be updating it to modify that file instead of the database but regardless it works and you can have it up and running in two commands.