ZoneMinder/zoneminder #869: Replay Event Stuck

mattionline, 2015-06-04:
Hello,

i have problems when i want to watch the events.
The most time it stucks at the beginning of the event, sometimes in the middle.
When i press pause and play nothing happens.
My system is a ubuntu server with 8gb ram and a quad core cpu.

mattionline, 2015-06-04:
I also have problems with the connection to the wlan cameras.
I have actually 3 repeaters and the wifi signal is very high and the pings are very good.
It seems that zoneminder cannot get the images of the 5 hd cameras so fast.
Do you have a solution for this 2 problems?

Logs: http://pastebin.com/m3zkjWmY

connortechnology, 2015-06-04:
Pings are all well and good, but what about bandwidth?  HD cameras require a fair amount.

Your use of the work repeater makes me think that your network is not as awesome as you think it is.

mattionline, 2015-06-04:
I need the 2 repeaters (300Mbit/s) because i have 3 floors in the house.
When i watch the camera directly there is no delay and everything is perfect, but in zoneminder it sometimes disconnect and shows up a red screen for a second.

Do you have a solution when i want to watch the captured events that it stucks/stops at the beginning or the middle of the event?

connortechnology, 2015-06-04:
This may be due to a flaw in the code that I have been fixing... 

How have you installed ZM? Which version?  Are you comfortable building from source?

mattionline, 2015-06-05:
I installed it directly from ubuntu with apt.
The version should be v1.28. In a few days i can look it up. Then im again at the house.
(https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder 15 weeks ago updated so it should be this version)
If the problem appears again ill build it directly from the source and try it again.
Thanks for your help. I will report in a few days if i solved the problem and how.
And i found a nice pdf document with some code edits to fix the problem with many monitors.
(http://dulisse.yolasite.com/resources/Zoneminder%20HOW%20TO%20-%20From%20scratch.docx)

connortechnology, 2015-06-05:
I might have to make a new ppa, with my latest developments in it.  Right now there is no one around to review my code and merge it. Let me see what I can do.

mattionline, 2015-06-15:
I have updated the server to the newest apt version. The version wasn't 1.28. It was 1.26.5.
Furthermore i have still the problem with 5 ip cameras. I posted the .docx howto from scratch but this didnt help. The cameras still disconnect for a few seconds and sometimes every camera disconnects. And this at least 3-4 times every hour. The connection to the camera itself is good with no delay (watched the camera through its own ip for a while) but i think zoneminder has a problem to handle 5 cameras.

The problem with the events is still there, that they stuck at one point. I tried it with different browsers (firefox, chrome) and with the smartphone (safari).
And sometimes it takes very long to load the livevideo. (server is connected per cable with a super speed) The load from my quadcore is between 0.5 and 1 so its not my server.

Can i optimize the bandwidth of the server in the zoneminder settings like in scratchs docx?
What can i do?
Could you update the ubuntu PPA?

SteveGilvarry, 2015-06-16:
Your logs seem to be full of connection related issues, both to the cameras and streaming. I know you can watch the one camera through it's interface but to test it you would need to watch all 5 cameras through their interface. Things to try 
1. Bring zm and cameras onto the same network device as a test. 
2. Upgrade to recent ZM, APT version is old.
3. Bear in mind some/most wifi extenders/repeaters will cut bandwidth in half if they link via wifi, running a repeater off a repeater might give real issues.
4. Lower resolution and frame rate on the camera see if that helps.

knnniggett, 2015-06-17:
@mattionline Using WiFi is not a reliable way to set up an IP camera system.  Additionally, you've compounded the problem by adding wireless repeaters.  Repeaters, by their very design, cut your bandwidth in half for each repeater the signal passes through: https://en.wikipedia.org/wiki/Wireless_repeater

This setup might work sufficiently for mobile devices, but it is not a reliable way to set up a device that runs 24/7 and constantly streams realtime traffic.  

This is guaranteed to be contributing to the problem.  When zoneminder bottlenecks somewhere, the symptoms are exactly what you have described.

Your network is the first thing that should be fixed.  Personally, I'd just pull a cat6 cable and be done with it.  However, I'm sure you've set up the WiFi because your perception is that you cannot accomplish that.

I've found a good compromise is to use Powerline Ethernet adapters.  That will give you a sufficiently fast and reliable, network connection to your zoneminder server.

If you have additional configuration or network issues you want to discuss, please take it to the user forums on zoneminder.com.  The github forum is intended for bug reporting and feature requests only.

closed 2015-06-17 by knnniggett