ZoneMinder/zoneminder #871: RTSP Buffer size too big

connortechnology, 2015-06-04:
On line 107 of zm_remote_camera_rtsp.cpp we initialize the buffer to width*height*colours.  This is a pretty large buffer, especially on HD cameras.  The data that this buff will hold will be compressed data.  

Later on, if more space is needed it will be resized.  All we are doing is setting the initial size. So it is preferable to use size that is close to the max that we will need..

We need some data.I can't help but think that 32k or 64k would be more reasonable. perhaps it should be configurable.

connortechnology, 2015-07-07:
So, after some changes to how reading from the socket is done, my buffers never seem to need to grow beyond 64k.  I chose 64k or in fact 2*ZM_NETWORK_BUFSIZ because the max read we will do when more than that is available is ZM_NETWORK_BUFSIZ.  So... to leave some room for previous data, etc I chose 2*ZM_NETWORK_BUFSIZ.  64k seems like a nice reasonable buffer.

liucougar, 2015-11-06:
@connortechnology any update on this please? should this be changed to reduce memory consumptions of high resolution monitors (HD+, like QXGA 2048x1536)?

connortechnology, 2015-11-06:
I've been using it here, and it works well.  Reduces mem use by several MB.  This issue affected the http code as well, and I think that fix has been merged.  These buffers grow if neccessary anyways.

closed 2017-03-18 by SteveGilvarry