ZoneMinder/zoneminder #872: Motion detection not working on 32bit system

seeebek, 2015-06-05:
I was trying to setup ZM for a friend. He has three Foscam Cameras: 2x FI9821w and 1x FI9805w. All are h264 cameras. 
The point is that everything works well on 64bit ubuntu (14.04 and 15.04) but motion detection does not on 32bit ubuntu (14.04 and 15.04). I'm not completely sure but I think it even doesn't work on ARM Ubuntu, however I did not test it after discovering the problem. The problem could be present on other distros.
Could some of you who own h264 cameras try running modect on 32bit systems to see if the problem is universal and reproducable?

I will try to investigate the reason for this problem but would be thankfull if some more experienced dev had a look into it.

connortechnology, 2015-06-05:
interesting observation, I will investigate.

Are you using ffmpeg, vlc or internal rtsp?

seeebek, 2015-06-05:
I used ffmpeg in this case. I think I also used vlc once with same effect. I'm not completely sure.

connortechnology, 2015-06-05:
There may be something to this.  Initial attempt is not detecting motion.

connortechnology, 2015-06-05:
Okay, so mine was caused by the directory for the monitor in events not being created.  This was causing zma to crash, and hence to events to be created.  Now I have events.

connortechnology, 2015-06-05:
check your logs, specifically zma

seeebek, 2015-06-05:
I will do it tomorrow. I'm travelling right now. 
Any idea why it is happening on 32bit and not on 64bit?

seeebek, 2015-06-05:
However I remember that I had no errors in logs only this common warning about shared memory. Additionally I could make forced alarm and it was saved as event. So this couldn't be the problem...

connortechnology, 2015-06-07:
You are going to have to crank up the logging level and do some debugging.  32bit motion detection is working just fine for me and likely for a lot of other people too.

davidsainty, 2015-06-08:
I might have an example of this too.  Just happened to happen today, after several weeks on this particular installation, and 8 days of up time.  I see a few Buffer overrun warnings (possibly relevant, but they don't seem out of place in the logs), then motion detection abruptly stopped.  Both zma and zmc continue to process, according to the logs, but no useful errors.  strace show both are at least actively handling data.

This is on a 32-bit ARM, but I don't doubt the 32-bitness is a red herring. ZoneMinder 1.28.1.

Restarting zoneminder fixed it.

Yeah, I realise this isn't a very helpful report :)

knnniggett, 2015-08-23:
I can confirm that motion detection is working fine on the following systems, running release 1.28.1:
32bit Fedora 22 on modern AMD cpu
Fedora 21 on Raspberry Pi 2 (arm7)

Given current information, I'm not sure this is a zoneminder problem.
I've got a Raspbeery Pi B+ lying around if you think it is worth testing on arm6 architecture.

In each case, I used an Airlink mjpeg ip camera at VGA resolution, and I used a Foscam 9821 h264 ip camera.

@davidsainty I think what you are describing is a different issue altogether.

knnniggett, 2016-04-09:
closing due to inactivity.

closed 2016-04-09 by knnniggett