ZoneMinder/zoneminder #878: cgi-bin -> symlink on zms not work for me (Gentoo/SystemD)

christophey2k, 2015-06-08:
---> /var/www/zm/cgi-bin 
lrwxrwxrwx 1 apache apache       3  9 juin  00:32 nph-zms -> zms
-rwxr-xr-x 1 apache apache 1426576  9 juin  00:31 zms

--> my web setting
PATH_ZMS 	--> Web path to zms streaming server (?) 	-->  /cgi-bin/nph-zms

Hello, all
I reinstall today from source zoneminder on my gentoo and my historical setting for
PATH_ZMS is /cgi-bin/nph-zms 

but on this fresh install, with this, haven't got any live stream.
I must be change to /cgi-bin/zms (not a problem)
but why this new feature (symlink) do not work on my gentoo ?

SteveGilvarry, 2015-06-10:
I am thinking this might be something like apache symlink settings, but I am guessing. Check your zoneminder.conf for +SymLinksIfOwnerMatch
ScriptAlias /cgi-bin/ "@CGI_PREFIX@"
    <Directory "@CGI_PREFIX@">
        Options +ExecCGI -MultiViews +SymLinksIfOwnerMatch
        AllowOverride All
    </Directory>

SteveGilvarry, 2015-06-16:
Closing out as likely this is OS or Apache issue not ZM, if you find more information that indicates otherwise please re-open. Thanks.

closed 2015-06-16 by SteveGilvarry