ZoneMinder/zoneminder #881: Merging angular-ui API branch into master

pliablepixels, 2015-06-10:
1) We need to merge the angular-ui API branch into master to get the latest APIs. 
2) Following this we will also need to merge https://github.com/ZoneMinder/ZoneMinder/pull/877

I'll start working on 1 for now.

connortechnology, 2015-06-11:
Um... I just looked ... and there is an api under master...I'm wondering how much is actually there already... looking...

pliablepixels, 2015-06-11:
Correct - the master does have an API branch but is not as updated as the angular-ui api branch - many APIs are not present

pliablepixels, 2015-06-13:
I just realized I did not merge in the following code into bootstrap.php 
```
Configure::write('ZM_CONFIG',  '@ZM_CONFIG@');
Configure::write('ZM_VERSION', '@VERSION@');
```
This is needed for the APIs to access config and version info. I'll do a PR

knnniggett, 2015-06-13:
After you checkout the branch, you need to do a "git submodule init"  and then possibly folloed by "git submodule update" to populate the submodule folder.

@connortechnology made some fixes to bootstrap.php and the related CMakeListFile so that those macros would get expanded properly.  Perhaps we just need to merge his commits into master.

pliablepixels, 2015-06-13:
I think he has already merged it to master -- the substitution in bootstrap is working.

I'm thinking of adding a version for the API that will be a minor to the version of ZM like this:
```
Configure::write('ZM_API_VERSION', '@VERSION@.1');
```
If ZM version is 1.28.1
This will make ZM_API_VERSION as 1.28.1.1

and in /api/getVersion.json, I can return both ZM_VERSION and ZM_API_VERSION

This will help in us making updates to the API incrementally. What do you think?

pliablepixels, 2015-06-13:
I've reworked this and put in an API version in cmakelists --> since that is where ZM version is set. Have done a PR

pliablepixels, 2015-06-14:
@connortechnology have you given any thought on authentication? Do you plan to link ZM portal login to API access or are you looking at an API auth. separate from ZM?

knnniggett, 2015-08-10:
Closing, since the API is now in master.

closed 2015-08-10 by knnniggett