ZoneMinder/zoneminder #889: 404 error on Ubuntu 14.04 install

cczus, 2015-06-17:
Fresh 14.04 Ubuntu install, downloaded the package from the conner PPA, install went fine.  When I try to load http://localhost/zm/ I get 404 error, http://localhost/zm/ loads the apache default page.  I have tried:

Adding a symbolic link to the zoneminder.conf file in /etc/apache2/sites-available/ as this was suggested for the same problem for deb.  

This is what I get when starting the services.

caseyz@ubuntuServer:~$ sudo service apache2 start
[sudo] password for caseyz:
 * Starting web server apache2                                                   *

caseyz@ubuntuServer:~$ sudo service zoneminder start
Starting ZoneMinder: success

Even though everything starts ok, still get the 404 error.

seeebek, 2015-06-18:
Try the following:
```sh
sudo ln -s /etc/zm/apache.conf /etc/apache2/conf-available/zoneminder.conf
sudo a2enconf zoneminder
```

knnniggett, 2015-06-18:
Please post which command resolves the issue.

I don't do zoneminder packaging on the debian side of things, but it sounds reasonable we could ask @connortechnology to add a few commands to the post-installation script if it turns out we are missing something simple.

cczus, 2015-06-18:
Thanks for the quick response.  Seebek's response resolved the issue.

seeebek, 2015-06-18:
In my opinion this should be included in the post-installation scripts. But it removes the freedom of custom apache configuration (might cause problems on some setups). 

The debian 8 script is much more sophisticated since it asks the user if he wants to perform those steps during installation. So the ubuntu packages should be modified to deb-conf like the debian 8 script. 

I was working on it but did not have the time lately. I still plan on doing it in the future.

connortechnology, 2015-06-18:
I have added a packaging directory for ubuntu1504, using the debian8 split packages approach, built using cmake.  I will be using it for ubuntu packaging going forward.

seeebek, 2015-06-18:
Great news.
I guess this issue can be closed?

closed 2015-06-19 by SteveGilvarry