ZoneMinder/zoneminder #898: Add isRunning field to States

pliablepixels, 2015-06-21:
As of today, there is no way to know which run-state ZM is currently in. Each run-state comprises of a specific configuration of monitors. zmpkg.pl switches to a run state but does not store the active state.
This makes it impossible to write cron scripts where two or more states have the same monitor configurations (example, out-of-house and vacation --> two different situations with the same monitor configs)

The proposal is to 

1) add an isRunning tinyint field to States which gets set to 1 when zmpkg changes state
2) Display the run state in ZM console at the top

@knnniggett -- do I need to modify zmupdate.pl as well in addition to modifying zm_create.sql and zm_update-1.28.99.sql?

knnniggett, 2015-06-21:
No mods to zmupdate.pl necessary.
Just add the schema change to the newest zm_updatex.xx.sql script and zmupdatw.pl will find and apply it  automatically. The sale must be written such that it can be run more than once without causing an error. See the existing scripts under the dB folder for examples.

pliablepixels, 2015-06-21:
Interestingly all past examples of ALTER TABLE don't bother checking if it
was already altered. I guess it will throw an error anyway. It's a little
convoluted in MySQL to check for existence of a column, so I guess I'll
just follow past precedence and just issue ALTER TABLE. if it was already
altered trying it again will not  add a duplicate column .

knnniggett, 2015-06-21:
Sometimes you don't have to when using Alter Table, but when you are trying to add a column you do.

See this example that adds the ExtendAlarmFrames column to the Zones table:
https://github.com/ZoneMinder/ZoneMinder/blob/master/db/zm_update-1.27.99.0.sql#L112

pliablepixels, 2015-06-21:
Okay done -- I've done a column check. Note that even if you don't it won't let you add the column (duplicate column error) - this is what update scripts were doing pre 1.22 and previous to it. I've however done it the proper way as you've recommended

knnniggett, 2015-06-21:
That's correct the older versions of ZoneMinder, prior to ~1.26, didn't implement sql updates the way we do now.

We made a design change recently to implement SQL such that it doesn't cause zmupdate to crash if it is run multiple times (which is often required during development).  Since the particular circumstances that cause zmupdate to crash only appear during development, between releases, it is in your best interest to do this.  

Older versions of ZoneMinder were developed, slowly, via 1-2 people, so it was not such an issue back then.

closed 2015-06-28 by pliablepixels