ZoneMinder/zoneminder #901: Allow use of UNIX File Sockets for MySQL Connection

josh4trunks, 2015-06-23:
Documentation for PDO is here.
http://php.net/manual/en/ref.pdo-mysql.connection.php

The $dbConn line needs to modified to allow inclusion of "unix_socket", and probably needs a bit of logic added to know when to include it.
ZoneMinder/web/includes/database.php

I'm not sure what else needs is crucial that needs modification for this to work. Most of the other instances of the database parameters being used seemed like distro specific stuff.

knnniggett, 2015-06-23:
Why?

It's working fine the way it is.  From the documentation you referenced:
```
 unix_socket

    The MySQL Unix socket (shouldn't be used with host or port).
```
We are specifying the host, so unix socket shouldn't be used:
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/database.php#L34

josh4trunks, 2015-06-23:
Im asking that the option be added, not that force admins to use either method. file sockets have higher performance because that aren't going through the network stack (being chunked, TCP reliability which is irrelevant on a local system)

It shouldn't be too hard to implement. I helped did this for owncloud and it when I have more time can try making a PR for zoneminder.

josh4trunks, 2015-06-23:
It should only take a line of logic to see if ZM_DB_SOCK ( or whatever ) is defined an to know what parameters to pass to PDO.

knnniggett, 2015-06-23:
Feel free to submit a pr.  Naturally, we want to make sure it doesn't break installations with a remote dB.

josh4trunks, 2015-06-23:
OK, will look into it sometime in the future.

knnniggett, 2015-08-09:
I'm closing this out due to inactivity.  The idea is sound.  If you choose to implement this, feel free to reopen this issue and generate a pull request.

josh4trunks, 2016-04-29:
I might try tackling this, this weekend.

It looks like zoneminder doesn't even take into account port numbers? so if someone tried connecting to a mysql running on a different port it wouldn't work?

josh4trunks, 2016-04-30:
@knnniggett I don't see a way to reopen this issue, can you do that?

Can someone please help me a little?
https://github.com/josh4trunks/ZoneMinder/tree/mysql_socket

I got the PHP done, but am having trouble with the Perl connection, at least I get an error when trying to start the zoneminder daemon.
It looks like `lib/ZoneMinder/Config.pm`, also has a db connection. Is there any possibilty to have this use `Database.pm` instead of having to have the database connection logic in 2 places for perl?
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Config.pm.in#L86-90

Thanks

josh4trunks, 2016-04-30:
I looks like Perl has connections made in multiple scripts, I got another error with `Logger.pm`.
I'll just put the needed logic everywhere, and let someone else decide if they want to made these more maintainable by just using `Database.pm`.

PR #1498 (2016-08-06): Add support for MySQL Port / Unix Socket
Logic is added to handle specified port / unix file socket for database connections.
For example, the following now all work as expected.
```
ZM_DB_HOST=localhost
ZM_DB_HOST=REMOTE_HOST:3307
ZM_DB_HOST=localhost:/path/to/non-default.sock
```

Fixes #901
closed 2016-08-06 by knnniggett