ZoneMinder/zoneminder #903: Logging after RTSP warning causes crash (segfault)

bill-mcgonigle, 2015-06-24:
I'm connecting to a network camera via RTSP (plays OK in mplayer) and it works for a few seconds before zm processes begin to segfault.

I'm not positive but it looks like the crash is in the sql logging part.  This is an example from the logfile:

````
Jun 24 02:20:26 zoneminder zmc_m1[1312]: WAR [Unexpected format RTSP interleaved data, resyncing by 48 bytes]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: WAR [48: 36 c0 a7 36 13 0a c1 91 bf e8 d5 c3 74 c7 06 66 ad a5 f6 64 25 44 fb 4f 9d c6 fa f5 4b 4f a5 0e 91 62 4f a2 9e 38 4e 81 ed d5 ea e7 27 d6 4e 7e]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: WAR [Unexpected format RTSP interleaved data, resyncing by 253 bytes]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: WAR [253: d7 b0 ee ce 7c e2 3c c8 0b a7 d2 18 46 30 77 00 41 b6 77 0f c5 4e f2 89 90 92 c6 9a dc 11 40 4f 8c 46 5a 2b 7d ea a4 c0 b3 0a b5 df 3d 2c b5 e2 c6 7b 49 bf a2 ea 26 62 d9 6a 0c ca 62 70 78 20 c2 e3 b3 2b 8e 78 34 42 6b 2c e8 03 d2 96 95 9e ac 8e b0 a5 11 a3 eb a4 60 32 30 de af 42 2a e6 af 6d e9 35 8e de 5b af 1c 9f aa 6d 0c 50 56 c2 a3 60 7f 22 0b 0f c8 9d 93 f6 ce 09 57 4b 42 03 8e 3b 04 23 1c 43 4d 28 0a de ac 61 6e 3b d5 b2 15 5f 22 f0 8f bb 58 0f 74 08 ef 67 9b 18 4c f7 c4 e5 26 45 c9 49 49 5e 73 3d ab 9a d2 f0 66 f9 d9 d3 07 e5 47 ac 96 c9 5b e9 ea f6 c9 3b 8f 63 b4 db bd 2c 8d ca 1c 4c 21 db 26 ca 16 f3 3f 52 53 25 c4 40 6e f0 f1 88 21 ec c6 c7 af a7 d6 36 76 e2 1a 3d 68 a8 93 2c 4a 80 3a 71 6c c6 41 b7 00 be 77 90 da 8f e3 b5 4e 04 13 01 b5]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Unexpected channel selector 170 in RTSP interleaved data]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: WAR [Unexpected format RTSP interleaved data, resyncing by 607 bytes]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: FAT [Can't insert log entry: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near ''607: 29 b5 bf 60 a8 77 41 03 3a 99 6e 2e 99 ee dd ec 04 b0 dd 6d 23 d9 21 40 07' at line 1]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Got signal 11 (Segmentation fault), crashing]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Signal address is 0x18, from 0x7f2734ef8c00]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 0: /usr/bin/zmc(_Z14zm_die_handleriP9siginfo_tPv+0x72) [0x7f27360e4332]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 1: /lib64/libpthread.so.0(+0xf130) [0x7f2734efe130]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 2: /lib64/libpthread.so.0(pthread_mutex_lock+0) [0x7f2734ef8c00]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 3: /lib64/libtspi.so.1(host_table_final+0x14) [0x7f272f4cbbb4]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 4: /lib64/ld-linux-x86-64.so.2(+0xfb5a) [0x7f2735e3eb5a]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 5: /lib64/libc.so.6(+0x38e49) [0x7f2731dbce49]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 6: /lib64/libc.so.6(+0x38e95) [0x7f2731dbce95]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 7: /usr/bin/zmc(_ZN6Logger8logPrintEbPKciiS1_z+0x42a) [0x7f273607bf9a]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 8: /usr/bin/zmc(_ZN6Logger8logPrintEbPKciiS1_z+0x962) [0x7f273607c4d2]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 9: /usr/bin/zmc(_ZN10RtspThread3runEv+0x4483) [0x7f27360cd3d3]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 10: /usr/bin/zmc(_ZN6Thread11mThreadFuncEPv+0xb2) [0x7f27360edb02]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 11: /lib64/libpthread.so.0(+0x7df5) [0x7f2734ef6df5]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: ERR [Backtrace 12: /lib64/libc.so.6(clone+0x6d) [0x7f2731e7a1ad]]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: INF [Backtrace complete, please execute the following command for more information]
Jun 24 02:20:26 zoneminder zmc_m1[1312]: INF [addr2line -e /usr/bin/zmc 0x7f27360e4332 0x7f2734efe130 0x7f2734ef8c00 0x7f272f4cbbb4 0x7f2735e3eb5a 0x7f2731dbce49 0x7f2731dbce95 0x7f273607bf9a 0x7f273607c4d2 0x7f27360cd3d3 0x7f27360edb02 0x7f2734ef6df5 0x7f2731e7a1ad]
Jun 24 02:20:26 zoneminder zmdc[1243]: ERR ['zmc -m 1' exited abnormally, exit status 11]
````

I rebuilt the zmrepo packages locally to get debug symbols, but if addr2line can use separate debuginfo, gnu hides that well (anybody know if/how?).  I ran zmc under gdb to get some info (local ip's anonymized):

````
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe5bf6700 (LWP 2160)]
0x00007ffff6ea4c00 in pthread_mutex_lock () from /lib64/libpthread.so.0
(gdb) thread apply all bt full

Thread 3 (Thread 0x7fffe5bf6700 (LWP 2160)):
#0  0x00007ffff6ea4c00 in pthread_mutex_lock () from /lib64/libpthread.so.0
No symbol table info available.
#1  0x00007ffff1477bb4 in host_table_final () from /lib64/libtspi.so.1
No symbol table info available.
#2  0x00007ffff7deab5a in _dl_fini () from /lib64/ld-linux-x86-64.so.2
No symbol table info available.
#3  0x00007ffff3d68e49 in __run_exit_handlers () from /lib64/libc.so.6
No symbol table info available.
#4  0x00007ffff3d68e95 in exit () from /lib64/libc.so.6
No symbol table info available.
#5  0x000055555557cf9a in Logger::logPrint (this=<optimized out>, hex=hex@entry=false, file=file@entry=0x5555556744a0 "/root/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_logger.cpp", 
    line=line@entry=588, level=level@entry=-3, fstring=fstring@entry=0x555555674394 "Can't insert log entry: %s") at /usr/src/debug/ZoneMinder-1.28.1/src/zm_logger.cpp:603
        classString = "FAT"
        timePtr = <optimized out>
        tid = <optimized out>
        logPtr = <optimized out>
        syslogStart = <optimized out>
        logString = "06/24/15 02:28:52.428829 zmc_m1[2160].FAT-/root/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_logger.cpp/588 [Can't insert log entry: You have an error in your SQL syntax; check the manual that corresponds "...
        timeString = "06/24/15 02:28:52.428829", '\000' <repeats 39 times>
        argPtr = {{gp_offset = 48, fp_offset = 48, overflow_arg_area = 0x7fffe5bf20f8, reg_save_area = 0x7fffe5bf2000}}
        timeVal = {tv_sec = 1435127332, tv_usec = 428829}
        syslogEnd = <optimized out>
#6  0x000055555557d4d2 in Logger::logPrint (this=<optimized out>, hex=hex@entry=true, file=file@entry=0x555555679698 "/root/rpmbuild/BUILD/ZoneMinder-1.28.1/src/zm_rtsp.cpp", 
    line=line@entry=752, level=level@entry=-1, fstring=fstring@entry=0x555555678b4d "%p (%d)") at /usr/src/debug/ZoneMinder-1.28.1/src/zm_logger.cpp:588
        sql = "insert into Logs ( TimeKey, Component, Pid, Level, Code, Message, File, Line ) values ( 1435127332.428243, 'zmc_m1', 2160, -1, 'WAR', '347: 91 9c 1a 95 85 6a f9 d2 45 9d ab 4b 18 54 fd 86 1a f4 5f b0 "...
        escapedString = 0x7fffe5bf2130 "347: 91 9c 1a 95 85 6a f9 d2 45 9d ab 4b 18 54 fd 86 1a f4 5f b0 9f 5a 5a 93 86 fb 97 31 aa 64 bd b5 43 30 8e 5d f1 e9 98 2c 1a 56 ee 35 ef 7e cf 81 9e e1 84 49 76 c0 77 88 37 de 97 29 f9 64 8c c1 41 "...
        classString = "WAR"
        timePtr = <optimized out>
        tid = <optimized out>
[... 89 lines cut ...]
        monitor_id = <optimized out>
        next_delays = 0x55555ad232f0
        now = {tv_sec = 1435127332, tv_usec = 418925}
        modes = <optimized out>
        host = <optimized out>
        alarm_capture_delays = 0x55555ad232d0
        delta_time = {positive = true, delta = <optimized out>, sec = <optimized out>, fsec = <optimized out>, prec = <optimized out>}
        file = <optimized out>
        long_options = {{name = 0x5555556739a1 "device", has_arg = 1, flag = 0x0, val = 100}, {name = 0x5555556739a8 "protocol", has_arg = 1, flag = 0x0, val = 114}, {
            name = 0x555555677dd2 "host", has_arg = 1, flag = 0x0, val = 72}, {name = 0x5555556739b1 "port", has_arg = 1, flag = 0x0, val = 80}, {name = 0x555555674c5a "path", 
            has_arg = 1, flag = 0x0, val = 112}, {name = 0x5555556739b6 "file", has_arg = 1, flag = 0x0, val = 102}, {name = 0x5555556739bb "monitor", has_arg = 1, flag = 0x0, val = 109}, 
          {name = 0x5555556739c3 "help", has_arg = 0, flag = 0x0, val = 104}, {name = 0x0, has_arg = 0, flag = 0x0, val = 0}}
        monitors = 0x55555ad01f80
        block_set = {__val = {2560, 0 <repeats 15 times>}}
````

When it crashes, mysql logs incomplete SQL statements (see middle one below).  I notice the size of the hex blob is pretty big, but on the mysql side it's a TEXT field, so that should be OK.  Maybe something is too small on the logging side [... cut ...]  7c bf 46 d0 87 b8 fb 26 7e 8e 49 67 77 5f b1 82 b4 ac f3 98 ca 94 18 26 e6 00 a4 db ac 5f 60 73 0e db 01 c0 16 6f a0 0b 7b fd 41 7b c4 b8 5e 6a 4
                   48 Query     insert into Logs ( TimeKey, Component, Pid, Level, Code, Message, File, Line ) values ( '1435116860.24626', 'zmdc', '6015', '-2', 'ERR', '\'zmc -m 1\' exited abnormally, exit status 11', 'zmdc.pl', NULL )
````

I think there's a separate RTSP handling problem here too, but that'll be easier to debug with stable logging.

connortechnology, 2015-06-24:
I wonder if this is a multi-threading concurrency issue, like more than one thread trying to access the db at once?

I've seen this, and in master we merged a patch to make db logging failures not fatal, but still there is something going on here.

bill-mcgonigle, 2015-06-24:
just for grins, based on [stackoverflow] (http://stackoverflow.com/questions/1455190/how-to-access-mysql-from-multiple-threads-concurrently) I added this to zmc.cpp:

````
        if ( mysql_library_init(0, NULL, NULL) ) 
        {
                fprintf( stderr, "Could not initialize MySQL library\n" );
                exit( -1 );
        }
````

and it didn't fix it.  Total shot in the dark (I'm out of my element); just adding here for completeness.  A link from the top comment there has guidance on thread safety in the mysql client library.

closed 2018-06-22 by stale[bot]