ZoneMinder/zoneminder #904: Can't compile Zoneminder v1.28.1

Traeld, 2015-06-24:
Hello,
i have installed all dependencies, but i am getting some errors when i try to build it.
I am using Raspberian at a Raspberry Pi 2.

Those are the error's i get:
```
/usr/bin/ld: libzm.a(zm_mpeg.cpp.o): undefined reference to symbol 'clock_gettime@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/librt.so.1: error adding symbols: DSO missing from command line
collect2: ld returned 1 exit status
src/CMakeFiles/nph-zms.dir/build.make:102: recipe for target 'src/nph-zms' failed
make[2]: *** [src/nph-zms] Error 1
CMakeFiles/Makefile2:120: recipe for target 'src/CMakeFiles/nph-zms.dir/all' failed
make[1]: *** [src/CMakeFiles/nph-zms.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2
```

Cmake was successful:
```
pi@raspberrypi ~/Downloads/ZoneMinder-1.28.1 $ sudo cmake .
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for ucontext.h
-- Looking for ucontext.h - found
-- Looking for sys/sendfile.h
-- Looking for sys/sendfile.h - found
-- Looking for sys/syscall.h
-- Looking for sys/syscall.h - found
-- Looking for syscall
-- Looking for syscall - found
-- Looking for sendfile
-- Looking for sendfile - found
-- Looking for backtrace
-- Looking for backtrace - found
-- Looking for backtrace_symbols
-- Looking for backtrace_symbols - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of siginfo_t
-- Check size of siginfo_t - done
-- Check size of ucontext_t
-- Check size of ucontext_t - done
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.7") 
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.26.0") 
-- Looking for curl/curl.h
-- Looking for curl/curl.h - found
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so  
-- Looking for include files stdio.h, jpeglib.h
[... 253 lines cut ...]
[ 71%] Building CXX object src/CMakeFiles/zm.dir/zm_stream.cpp.o
[ 73%] Building CXX object src/CMakeFiles/zm.dir/zm_thread.cpp.o
[ 75%] Building CXX object src/CMakeFiles/zm.dir/zm_time.cpp.o
[ 77%] Building CXX object src/CMakeFiles/zm.dir/zm_timer.cpp.o
[ 79%] Building CXX object src/CMakeFiles/zm.dir/zm_user.cpp.o
[ 81%] Building CXX object src/CMakeFiles/zm.dir/zm_utils.cpp.o
[ 83%] Building CXX object src/CMakeFiles/zm.dir/zm_zone.cpp.o
Linking CXX static library libzm.a
[ 83%] Built target zm
Scanning dependencies of target nph-zms
[ 85%] Building CXX object src/CMakeFiles/nph-zms.dir/zms.cpp.o
Linking CXX executable nph-zms
/usr/bin/ld: libzm.a(zm_mpeg.cpp.o): undefined reference to symbol 'clock_gettime@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/librt.so.1: error adding symbols: DSO missing from command line
collect2: ld returned 1 exit status
src/CMakeFiles/nph-zms.dir/build.make:102: recipe for target 'src/nph-zms' failed
make[2]: *** [src/nph-zms] Error 1
CMakeFiles/Makefile2:120: recipe for target 'src/CMakeFiles/nph-zms.dir/all' failed
make[1]: *** [src/CMakeFiles/nph-zms.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2

```

I've searched alot, but did not find a solution, can you help me please?

knnniggett, 2015-06-24:
According to this:
https://www.raspberrypi.org/forums/viewtopic.php?t=19756

Raspberian needs the "rt" linker flag set.

You could try adding that here:
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/CMakeLists.txt#L23

But, I would rather you first try setting the ZM_EXTRA_LIBS flag at build time like so:
`cmake -DZM_EXTRA_LIBS="rt" .`

or you might need to try it like so:
`cmake -DZM_EXTRA_LIBS="-lrt" .`

Please report back which works.

Traeld, 2015-06-24:
Thanks alot for the fast reply and my problem's solution!

I tried both commands seperatly, both Flags "rt" and "-lrt" are working.

closed 2015-06-24 by knnniggett