ZoneMinder/zoneminder #905: Track Solaris Support

knnniggett, 2015-06-25:
This issue has been created as the central discussion point for support for Solaris.  Let's keep the conversation here, rather than in the individual pull requests.

knnniggett, 2015-06-25:
After researching our options, I opted to leverage cmakes ability to find different names of the same package via a single command.  The result is a much cleaner approach.

@whorfin can you test #906 and post your results?  Please post your build output.  I'm not looking for a new pull request at this time.

knnniggett, 2015-08-10:
I'm closing this due to inactivity.
If @whorfin or anyone else desires to continue to work on this, then let me know and I will reopen.

whorfin, 2015-08-23:
My apologies, I've been extremely busy and out of town.
After re-merging everything upstream back to my tree, I followed instructions and am testing #906 
As an aside - it appears that 
web/api/app/Config/core.php
is checked in to github, whereas as a dependent file [processed from core.php.default] it seems it should not be.
Here is the build output below.

Note that, as indicated in my patch/notes on the broken CheckSendfile.cmake, HAVE_SENDFILE4_SUPPORT fails

At the moment, I have a patch for CMakeLists.txt to fix this, but I understand you don't want
pulls at the moment.  I confined changes to that file and am not touching cmake/Modules

$ cmake \
>     -DZM_WEBDIR="/opt/local/share/httpd/htdocs/zm" \
>     -DZM_CGIDIR="/opt/local/libexec/cgi-bin" \
>     -DZM_WEB_USER="www" \
>     -DZM_WEB_GROUP="www" \
>     -DZM_EXTRA_LIBS="socket;nsl;sendfile" \
> .
-- Performing Test HAVE_SENDFILE4_SUPPORT
-- Performing Test HAVE_SENDFILE4_SUPPORT - Failed
-- Performing Test HAVE_SENDFILE7_SUPPORT
-- Performing Test HAVE_SENDFILE7_SUPPORT - Failed
-- Performing Test HAVE_SENDFILE6_SUPPORT
-- Performing Test HAVE_SENDFILE6_SUPPORT - Failed
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - not found
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - not found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for ucontext.h
-- Looking for ucontext.h - found
-- Looking for sys/sendfile.h
-- Looking for sys/sendfile.h - found
-- Looking for sys/syscall.h
-- Looking for sys/syscall.h - found
-- Looking for syscall
-- Looking for syscall - found
-- Looking for sendfile
-- Looking for sendfile - not found
-- Looking for backtrace
-- Looking for backtrace - found
-- Looking for backtrace_symbols
-- Looking for backtrace_symbols - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of siginfo_t
-- Check size of siginfo_t - done
-- Check size of ucontext_t
-- Check size of ucontext_t - failed
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.8")
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Found CURL: /usr/lib/libcurl.so (found version "7.42.1")
-- Looking for curl/curl.h
-- Looking for curl/curl.h - found
-- Found JPEG: /opt/local/lib/libjpeg.so
-- Looking for include files stdio.h, jpeglib.h
-- Looking for include files stdio.h, jpeglib.h - found
[... 132 lines cut ...]
      /opt/local/lib
    runtime library [libpcre.so.1] in /usr/lib may be hidden by files in:
      /opt/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at src/CMakeLists.txt:14 (add_executable):
  Cannot generate a safe runtime search path for target zmu because files in
  some directories may conflict with libraries in implicit directories:

    runtime library [libz.so] in /usr/lib may be hidden by files in:
      /opt/local/lib
    runtime library [libcurl.so.4] in /usr/lib may be hidden by files in:
      /opt/local/lib
    runtime library [libssl.so.1.0.0] in /usr/lib may be hidden by files in:
      /opt/local/lib
    runtime library [libcrypto.so.1.0.0] in /usr/lib may be hidden by files in:
      /opt/local/lib
    runtime library [libpcre.so.1] in /usr/lib may be hidden by files in:
      /opt/local/lib

  Some of these libraries may not be found correctly.

-- Generating done
-

whorfin, 2015-09-06:
Another issue:
Paths seem to remain hardcoded; mysql and mysqldump [for example] are in my path when i
build, but the build system doesn't notice my environment and instead uses these hard coded paths in [for example] zmupdate.pl.in and zmcamtool.pl.in
$ENV{PATH}  = '/bin:/usr/bin:/usr/local/bin';  

On OmniOS with pkgsrc, mysql and friends [and much other goodies] live in
/opt/local/bin

The build system should resolve all needed executables, and use those paths in the scripts rather than assuming standard paths.

whorfin, 2015-09-06:
One more; i suspect this isn't Solaris related, but while attempting to bringup a system from my last build [1.28.1] to the current master, zmupdate.pl fails in a way which looks suspiciously as if
updates are being applied in alphabetical as opposed to numerical order.  .99 should surely
not be applied after .104...

$ sudo zmupdate.pl

Initiating database upgrade to version 1.28.104 from version 1.28.1

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop :

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /var/tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.28.104
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.1 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.100 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.100.
Upgrading DB to 1.28.101 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.101.
Upgrading DB to 1.28.102 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.102.
Upgrading DB to 1.28.103 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.103.
Upgrading DB to 1.28.104 from 1.28.1
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.104.
Upgrading DB to 1.28.99 from 1.28.1
Warning: Using a password on the command line interface can be insecure.
ERROR 1142 (42000) at line 376: CREATE command denied to user 'zmuser'@'localhost' for table 'Servers'
Output: Column Id exists in States
Column Id exists in States
Column IsActive  exists in States
Column IsActive  exists in States

whorfin, 2015-09-06:
There is a typo in zmlinkcontent.sh.in on line 72; a missing space between the quoted filename and closing square bracket.

whorfin, 2015-09-06:
Fair warning:
    -DCMAKE_SKIP_RPATH:BOOL=YES 
is needed as an argument to cmake, or all RPATH/RUNPATH are scrubbed during installation, rendering -R arguments to the linker very confusing - the build tree version will run, the install will not.
this is an extra fun "feature" of cmake

whorfin, 2015-09-06:
More cmake fun - I didn't at first realize all my ffmpeg/libavformat/libavcodec was being missed, as new names are being displayed such as AVFormat

The cmake install system is not finding any of my goodies, though i'm passing CMAKE_LIBRARY_PATH and CMAKE_INCLUDE_PATH entries and perhaps more importantly for standard builds, "pkg-config" know where there are.
pkg-config finds them, cmake fails

whorfin, 2015-09-06:
Comment Error:
INSTALL says:

CMAKE_PREFIX_PATH - Use this to add to both include and library search paths

This appears to not be true; if CMAKE_PREFIX_PATH is used, it must also contain "/usr" or I get failure to find various perl modules

whorfin, 2015-09-06:
After all this, all camera feeds are broken, even with simple fallbacks.
<sigh>
zm_remote_camera_rtsp.cpp failing at line 310/311

Linwood-F, 2015-09-06:
Re alphabetical order -- that came up in someone on ... IRC?   Can't recall, but someone who was trying to upgrade.  The problem (with the problem) was I think it cumulatively worked anyway so I do not think anyone actually fixed it. but maybe this discussion will spur someone's memory.

knnniggett, 2015-09-06:
Guys, I created this thread to keep it on topic with Solaris Support so I wouldn't have to wade trough irrelevant discussion and commentary to get to what mattered.  Let's stay on topic!  Save the discussion for our irc channel.

@whorfin I'm going to enjoy the rest of my holiday weekend.  When I get back, I'm going to focus solely on your post from 15 days ago, and I'm going to ignore all your other posts.  I know you want to get this fixed, but spamming our forum with a bunch of errors, without logically connecting them, only shows you are digging yourself into a deeper hole. That only makes it harder to troubleshoot. So back it up. We are going to focus on getting cmake going, and are not going to worry about anything else until that is solved.

whorfin, 2015-09-06:
Totally understood, I include these things for context, since the cmake change came with a lot of other changes.
While I have a build which builds and an install which installs, and now runs, my cameras don't work.
I don't know how to be certain that the cmake changes didn't cause this.
Fair enough?

To back it all the way up, here are the build commands I am now using.
I no longer use -L and -R entries coded into CXX/CFLAGS, as i am trying to be more cmakeish.
As noted, if you use -R you will be required to add 
-DCMAKE_SKIP_RPATH:BOOL=YES 
but obvs don't use that with the below

export PKG_CONFIG_PATH=/usr/lib/pkgconfig/:/opt/local/lib/ffmpeg2/pkgconfig:/opt/local/lib/pkgconfig
export CXXFLAGS=-m64
export CFLAGS=-m64
export PERL=/usr/perl5/5.16.1/bin/amd64/perl
export ZM_SSL_LIB=openssl # passing this as a -D to cmake does not work
cmake \
    -DZM_WEBDIR="/opt/local/share/httpd/htdocs/zm" \
    -DZM_CGIDIR="/opt/local/libexec/cgi-bin" \
    -DZM_WEB_USER="www" \
    -DZM_WEB_GROUP="www" \
    -DZM_EXTRA_LIBS="socket;nsl;sendfile" \
    -DZM_CONTENTDIR="/scratch/zm" \
    -DCMAKE_INCLUDE_PATH="/opt/local/include/ffmpeg2" \
    -DCMAKE_LIBRARY_PATH="/opt/local/lib/ffmpeg2" \
    -DCMAKE_PREFIX_PATH="/usr:/opt/local" \
    -DCMAKE_BUILD_WITH_INSTALL_RPATH:BOOL=YES \
    -DCMAKE_INSTALL_RPATH_USE_LINK_PATH:BOOL=YES  \
.
make

Have an excellent holiday

knnniggett, 2015-09-21:
My apologies for not starting on this. I'm tracking down a problem with a particular camera that has developed in master, and it has me completely stymied.  Solaris support is on my priority list.

On a slightly different note, is there by chance any inexpensive hardware I could buy, which would allow me to create the same environment you are using?  I'm assuming Solaris is incompatible with x86 hardware.

whorfin, 2015-09-21:
I appreciate the update, thanks.  I'll be unable to contribute to this from the end of this month until mid-November, which is why I was death-marching in such a frenzy last month.

As to your second question, you'll be glad to know you don't need anything special.  I'm using "Solaris" in the generic sense, I am in fact running the Illumos branch in general, and OmniOS specifically, along with the joyent "pkgsrc" extensions for things like Apache/mysql/php/ffmpeg/avcodec/etc [as you can see from my cmake commands above; pkgsrc lives in /opt/local].  To be pedantically specific, perl is from OmniOS core, with extensions pulled from CPAN.  That should also be clear from my cmake config.

If you want to play along:
http://omnios.omniti.com/wiki.php/Installation
[use CurrentStable, not LTS or Bloody]
There are even Vagrant Box images, if you don't want to do a bare-metal install.

OmniOS is a bulletproof server environment, and ZFS/dtrace/crossbow/svcadm make the ecosystem uniquely awesome.

The only big gotcha is that non-64-bit hardware will see limits on maximum drive size support, and hardware support is more limited than windoze.

Cheers

knnniggett, 2015-10-24:
Until now, my understanding of Solaris was an OS that ran on big-endian hardware, much like IBM AIX.
I had no idea this ran on common hardware.

I've started familiarizing myself with the environment. Did you by chance write down the steps you took to install all of ZoneMinder dependencies? I'm running into some issues getting all the Perl modules to install, even when using cpan -f.

whorfin, 2015-10-25:
I did make notes on everything, but I'm out of the country, with crappy internet until next month.
One trick with perl modules is to be mindful of 64/32 bit versions of dependencies.  
I can send more info when I return

Cheers

whorfin, 2015-11-09:
Here's a shortened version of my notes on installing dependencies into a fresh OmniOS bare-metal install.

I'm using a 64-bit machine/OS, but note that lots of things default to 32-bit for various reasons.  I want 64-bit zm for simplicity with various 3rd party package compatibility.

"pkg" is the native OmniOS package manager
"pkgin" references an install of Joyent's builds of various 3rd party things, particularly the nightmare that is ffmpeg.  One can use some special OmniOS repos to get the perl dependencies, but my path is to use the OmniOS perl, and use CPAN to download/build/install needed perl packages.
Root, zone access or sudo etc are assumed where necessary.

Install joyent pkgsrc; more info here https://pkgsrc.joyent.com/install-on-illumos/
```
$ curl http://pkgsrc.joyent.com/packages/SmartOS/bootstrap/bootstrap-2015Q2-x86_64.tar.gz | sudo gtar -zxpf - -C /
```
you may or may not need some of the refreshing magics
```
$ pkg_admin rebuild
$ pkgin update
```

Finally be sure the right stuff is in your path.
For my installations I use `myOwnStuf:/usr/gnu/bin:/usr/bin:/usr/local/bin:/usr/sbin:/sbin:/opt/local/sbin:/opt/local/bin`

Install perl-64
```
$ pkg install perl-64
```

Depending on how you've got your install set up, you may find `perl --version` is 32-bit. We can fix this via an explicit path later.  You may also find that the auto-build bit-switching mechanism is busted, if it is, you need to rebuild a link. For example.
```
$ ls -l /usr/bin/perl
lrwxrwxrwx 1 root root 24 Dec 31 16:48 /usr/bin/perl -> ../perl5/5.16.1/bin/perl
$ cd /usr/bin/amd64
$ ln -s ../../perl5/5.16.1/bin/amd64/perl .
```

Now we want to install areasonably modern build chain, but we wish to get it from the
 OmniOS repo, not pkgin - don't worry, we can link pkgin objects with the native compiler.  YMMV if you use a different pkgin version, but it is easy enough to tell what base assumptions are once you get far enough in, with `ffmpeg2 --version` later.
```
$ pkg install gnu-make
$ pkg install libtool automake autoconf
$ pkg install developer/library/lint
$ pkg install system/header
$ pkg install header-math
$ pkg install gcc48
$ pkg install automake libtool
$ pkg install pkg-config
```

Now we'll use pkgin to get an AMP stack.  I've split these out for clarity but you can of course do all in one big bang.
```
$ pkgin install apache
$ pkgin install mysql-server mysql-client
$ pkgin install php ap24-php56 php56-pdo php56-pdo_mysql
```
I'll assume you know how to dink with the multitudinous config files of an AMP install.
Once you're good, fire up and enable the services
```
$ svcadm enable apache
$ svcadm enable mysql
```

If you see errors related to projadd failing with an invalid projid, just re-run the projadd line invoked by `/opt/local/pkg/mysql-server-BLAH/+INSTALL`, omitting the `-p 70` as projids 0-99 are reserved in Solaris.

`svcadm refresh apache` after changing any apache configs.  You'll need `restart` instead when changing modules, as everything needs to be reloaded.

Check status of services with `svcs -xv`

Time for all the various 3rd-party dependencies.
```
$ pkgin install cmake
$ pkgin install ffmpeg2
[... 142 lines cut ...]
 then fire it up again.  This is what I needed to install, found via grepping ^ac_perl_modules from zoneminder's configure.  You may want to script this, as this is "go make tea" time.
```
install DBI
notest install DBD::mysql	  # to avoid login fail if you've already locked down your mysql
install Getopt::Long
install Time::HiRes
install Date::Manip
install LWP::UserAgent		# crapload of dependencies, including TLS and SSL
install ExtUtils::MakeMaker
install Sys::Mmap
install Module::Load
install Device::SerialPort
install Net::FTP
install Net::SFTP::Foreign	# all of SFTP is huge crapton of math, SSH1/2, crypto etc
		# all the way to Net::SSH::Perl, probably can fix with dependency
install Expect
install Archive::Tar
install Archive::Zip
install Net::SMTP
install MIME::Lite
install MIME::Entity
install X10::ActiveHome
```

There ya go.  "It's just that simple."

closed 2017-03-18 by SteveGilvarry