ZoneMinder/zoneminder #911: RTSP fails when "/" at end of URL in DESCRIBE response header

bhuddah, 2015-06-27:
this only happens with "External" source type.
workaround: use ffmpeg or libvlc source type.

```
Jun 27 16:13:52 ubuntukas zmc_m12[14954]: INF [Recieved new Content-Base in DESCRIBE reponse header. Updated device Url to: 'rtsp://192.168.10.28/live1.sdp/']
```

knnniggett, 2015-06-27:
Do you mean a source type of "remote"?  there is no source type of "external".

In any case, with RTSP issues, we need to see a full level 5 debug showing the initial RTSP handshake leading all the way up to the error message.  This may be fixed already in master, but I can't tell until I can see the whole debug session.

bhuddah, 2015-06-27:
Hi,
yes, of course "remote" ... i was ad hoc translating from the german gui here.

i hope the attached output helps.

```
06/27/15 17:16:19.351833 zmc_m4[17733].DB1-zm_logger.cpp/230 [LogOpts: level=DB5/DB5, screen=OFF, database=FAT, logfile=DB5->/var/log/zm/zm_debug.log.17733, syslog=DB9]
06/27/15 17:16:19.352106 zmc_m4[17733].DB1-zm_utils.cpp/265 [Detected a x86\x86-64 processor with SSE3]
06/27/15 17:16:19.354098 zmc_m4[17733].DB1-zm_monitor.cpp/2554 [Got 1 monitors]
06/27/15 17:16:19.354329 zmc_m4[17733].DB1-zm_monitor.cpp/2589 [Got 1 for v4l_captures_per_frame]
06/27/15 17:16:19.354445 zmc_m4[17733].DB2-zm_camera.cpp/39 [New camera id: 4 width: 1600 height: 1200 colours: 3 subpixelorder: 6 capture: 1]
06/27/15 17:16:19.356896 zmc_m4[17733].DB1-zm_thread.cpp/241 [Creating thread]
06/27/15 17:16:19.357147 zmc_m4[17733].DB2-zm_rtsp.cpp/197 [RTSP Local SSRC is 5076c4b9]
06/27/15 17:16:19.357234 zmc_m4[17733].DB1-zm_thread.cpp/281 [Starting thread]
06/27/15 17:16:19.357477 zmc_m4[17736].DB2-zm_thread.cpp/253 [Invoking thread]
06/27/15 17:16:19.357717 zmc_m4[17733].DB1-zm_thread.cpp/302 [Started thread 17736]
06/27/15 17:16:19.358047 zmc_m4[17733].DB2-zm_image.cpp/167 [Blend: Using SSE2 fast blend function]
06/27/15 17:16:19.358143 zmc_m4[17733].DB2-zm_image.cpp/220 [Delta: Using SSE2 delta functions]
06/27/15 17:16:19.358242 zmc_m4[17733].DB2-zm_image.cpp/254 [Deinterlace: Using standard delta functions]
06/27/15 17:16:19.358308 zmc_m4[17733].DB2-zm_image.cpp/260 [Image buffer copy: Using SSE2 aligned memcpy]
06/27/15 17:16:19.358357 zmc_m4[17736].DB2-zm_rtsp.cpp/48 [Sending RTSP message: OPTIONS rtsp://restaurant:554/live2.sdp RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 1

]
06/27/15 17:16:19.358550 zmc_m4[17733].DB3-zm_image.cpp/267 [Setting up static colour tables]
06/27/15 17:16:19.358684 zmc_m4[17733].DB1-zm_monitor.cpp/379 [monitor purpose=1]
06/27/15 17:16:19.358749 zmc_m4[17733].DB1-zm_monitor.cpp/388 [mem.size=460806344]
06/27/15 17:16:19.358845 zmc_m4[17733].DB1-zm_monitor.cpp/537 [Unable to map file /dev/shm/zm.mmap.4 (460806344 bytes) to locked memory, trying unlocked]
06/27/15 17:16:19.358915 zmc_m4[17733].DB1-zm_monitor.cpp/539 [Mapped file /dev/shm/zm.mmap.4 (460806344 bytes) to locked memory, unlocked]
06/27/15 17:16:19.358982 zmc_m4[17733].DB3-zm_monitor.cpp/566 [Aligning shared memory images to the next 16 byte boundary]
06/27/15 17:16:19.365464 zmc_m4[17736].DB2-zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 200 OK
CSeq: 1
Date: Sat, Jun 27 2015 15:16:40 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

 (152 bytes)]
06/27/15 17:16:19.365607 zmc_m4[17736].DB2-zm_rtsp.cpp/48 [Sending RTSP message: DESCRIBE rtsp://restaurant:554/live2.sdp RTSP/1.0
User-Agent: ZoneMinder/1.28.1
CSeq: 2

]
06/27/15 17:16:19.966336 zmc_m4[17733].DB1-zm_monitor.cpp/447 [Monitor Restaurant has no zones, adding one.]
06/27/15 17:16:19.966505 zmc_m4[17733].DB1-zm_zone.cpp/51 [Initialised zone 0/All - 1 - 1600x1200 - Rgb:0000ff, CM:3, MnAT:15, MxAT:0, MnAP:50, MxAP:75000, FB:3x3, MnFP:50, MxFP:50000, MnBS:10, MxBS:0, MnB:0, MxB:0, OF: 0, AF: 0]
06/27/15 17:16:20.176746 zmc_m4[17733].DB1-zm_monitor.cpp/457 [Monitor Restaurant has function 3]
06/27/15 17:16:20.176837 zmc_m4[17733].DB1-zm_monitor.cpp/458 [Monitor Restaurant LBF = '%N - %Y-%m-%d %H:%M:%S.%f', LBX = 0, LBY = 0]
06/27/15 17:16:20.176854 zmc_m4[17733].DB1-zm_monitor.cpp/459 [Monitor Restaurant IBC = 80, WUC = 40, pEC = 1, PEC = 80, EAF = 1, FRI = 10000, RBP = 6, ARBP = 6, FM = 0]
06/27/15 17:16:20.179432 zmc_m4[17733].DB1-zm_zone.cpp/957 [Got 1 zones for monitor Restaurant]
06/27/15 17:16:20.179691 zmc_m4[17733].DB5-zm_zone.cpp/989 [Parsing polygon 0,0 1599,0 1599,1199 0,1199]
06/27/15 17:16:20.179759 zmc_m4[17733].DB3-zm_zone.cpp/823 [Parsing polygon string '0,0 1599,0 1599,1199 0,1199']
06/27/15 17:16:20.179883 zmc_m4[17733].DB3-zm_zone.cpp/861 [Got coordinate 0,0 from polygon string]
06/27/15 17:16:20.179947 zmc_m4[17733].DB3-zm_zone.cpp/861 [Got coordinate 1599,0 from polygon string]
06/27/15 17:16:20.180009 zmc_m4[17733].DB3-zm_zone.cpp/861 [Got coordinate 1599,1199 from polygon string]
06/27/15 17:16:20.180069 zmc_m4[17733].DB3-zm_zone.cpp/861 [Got coordinate 0,1199 from polygon string]
06/27/15 17:16:20.180174 zmc_m4[17733].DB3-zm_zone.cpp/881 [Successfully parsed polygon string]
06/27/15 17:16:20.180240 zmc_m4[17733].DB1-zm_zone.cpp/51 [Initialised zone 2/All - 1 - 1600x1200 - Rgb:0000ff, CM:3, MnAT:23, MxAT:0, MnAP:920, MxAP:736449, FB:3x3, MnFP:368, MxFP:0, MnBS:184, MxBS:0, MnB:1, MxB:0, OF: 0, AF: 0]
06/27/15 17:16:20.265501 zmc_m4[17733].DB1-zm_monitor.cpp/2850 [Loaded monitor 4(Restaurant), 1 zones]
06/27/15 17:16:20.265649 zmc_m4[17733].INF-zmc.cpp/208 [Starting Capture]
06/27/15 17:16:20.265726 zmc_m4[17733].DB2-zm_remote_camera_rtsp.cpp/149 [Waiting for sources]
06/27/15 17:16:20.365738 zmc_m4[17736].DB2-zm_rtsp.cpp/74 [Received RTSP response: RTSP/1.0 200 OK
CSeq: 2
Date: Sat, Jun 27 2015 15:16:40 GMT
Content-Base: rtsp://192.168.10.20/live2.sdp/
Content-Type: application/sdp
Content-Length: 647

v=0
o=- 1435406388601945 1 IN IP4 192.168.10.20
s=RTSP/RTP stream 2 from Margarete3
i=live2.sdp with v2.0
[... 160 lines cut ...]
06/27/15 17:16:20.888098 zmc_m4[17736].DB5-zm_rtp_data.cpp/44 [Ver: 2]
06/27/15 17:16:20.888115 zmc_m4[17736].DB5-zm_rtp_data.cpp/45 [P: 0]
06/27/15 17:16:20.888130 zmc_m4[17736].DB5-zm_rtp_data.cpp/46 [Pt: 96]
[... cut ...]
06/27/15 17:16:20.888609 zmc_m4[17736].DB5-zm_rtp_source.cpp/173 [Packet RTP time = 0]
06/27/15 17:16:20.888751 zmc_m4[17736].DB5-zm_rtp_source.cpp/175 [Packet transit RTP time = b4c6]
06/27/15 17:16:20.888768 zmc_m4[17736].DB5-zm_rtp_source.cpp/193 [RTP Jitter: 0]
06/27/15 17:16:20.888784 zmc_m4[17736].DB4-zm_rtsp.cpp/754 [Received]
06/27/15 17:16:20.888799 zmc_m4[17736].DB5-zm_rtsp.cpp/811 [sendkeepalibe 1, timeout 0, now: 1435418180 last: 1435418180 since: 0]
06/27/15 17:16:20.888824 zmc_m4[17736].DB4-zm_rtsp.cpp/732 [Read 4 bytes on sd 4, 4 total]
```

knnniggett, 2015-06-27:
Looks normal.  I don't see an error. However, there is only 2 seconds of data there.
We need to see all the data from the beginning, ending with the error you are experiencing, and everything in-between.  You might be better off posting it into a gist.

bhuddah, 2015-06-27:
hi,

i've created a gist with a complete debug log here:

https://gist.github.com/bhuddah/2e49e77a622c653093dd

knnniggett, 2015-06-27:
Well, that sucks.  Even Gist complains it is too big. :-(

knnniggett, 2015-06-27:
The information I was able to download from your log starts at 17:39:21.533100 and ends at 
17:39:21.547340. There is no error in that information either.  What exactly isn't working?  What error messages do you see?

The message in your original post is not an error.  It is just telling you that the camera thinks you should be using `rtsp://192.168.10.28/live1.sdp/` rather than what you entered into zoneminder's config: `rtsp://restaurant:554/live2.sdp`

bhuddah, 2015-06-27:
There is no error in the complete log. The Cam just stops working (at the time where there are only keepali**b**e messages in the log) (also there is a typo in all of these messages) at 06/27/15 17:40:26.016770

"restaurant" is an /etc/hosts alias for 192.168.10.28 both the ip and the hostname are working. the only possibility is that it's something related to the terminating "/" there.
please ignore the difference between live1.sdp and live2.sdp. the initial message was from a different cam than that debug log. i have this problem with multiple cams (all d-link).

knnniggett, 2015-06-27:
The keepalibe spelling typo has already been fixed in master.

So you are saying that the stream comes in fine initially, and then stops? After how long?

bhuddah, 2015-06-27:
i'd say "see the log" but... 
17:39:21.533100 to 17:40:26.016770 is about a minute. that seems to be the time frame.

knnniggett, 2015-06-27:
Then this has nothing to do with the "/" in the url.  Otherwise, it would never have worked at all.

I'm not sure there is anything we can do with this without some kind of hard data that says what is going wrong.

bhuddah, 2015-06-27:
so the cam actually uses the url from the DESCRIBE header right from the start?

knnniggett, 2015-06-27:
That url is coming from the camera.  The camera is saying "Hey, your initial url was close, but I want you to use this one instead".  This is part of the RTSP standard as described in the RTSP RFC.

bhuddah, 2015-06-27:
ok. i have no idea what is going on then. :(

knnniggett, 2015-06-27:
There has to be some kind of error, somewhere, if you are losing the image.
I'm sure you you've noticed the RTSP handshake messages: OPTIONS, DESCRIBE, PLAY, etc.

At the end of the session, you should see a message called TEARDOWN.  Check for an error just before that.

bhuddah, 2015-06-27:
the debug log stops mid-line before the actual end of the session.

knnniggett, 2016-02-19:
Please upgrade to 1.29.0 if you have not already and then disable the RTSP Describe response feature in the monitor config.

closed 2017-03-18 by SteveGilvarry